ra3xdh/qucs_s #189: Add variable sweep support using the alterparam command

ra3xdh, 2022-11-24:
Ngspice has added `alterparam` command in the recent versions https://ngspice.sourceforge.io/docs/ngspice-html-manual/manual.xhtml#magicparlabel-24765 This command allows to modify the variable defined in `.PARAM` section. This could be used to implement variable parameter sweet that is currently not supported by Qucs-S+Ngspice.

ra3xdh, 2022-11-27:
I have added a basic support for `alterparam` on`current` branch. This makes possible sweeping variables with Qucs-S+Ngspice. Qucs-S recognizes variables automatically and apply appropriate command. See attached screenshot for example.
[image]

0range0range, 2022-12-05:
I tried the same circuit as in the picture on Windows and I have a question regarding plotting.
If I sweep parameter by name, then when I plot multiple curves for each parameter valuer appear. 
But, if I sweep something defined in `.param`, then although resulting datasets are the same, when I select to plot `v(out)` for example - only curve corresponding to the first parameter value appears on the plot.

How to plot for multiple parameter values, defined with `.param`?

ra3xdh, 2023-02-02:
Closing this. The feature expected in the upcoming v1.0.1 release.

closed 2023-02-02 by ra3xdh