ra3xdh/qucs_s #190: Qt6

ra3xdh, 2022-11-25:
It's need to check if the Qucs-S could be compiled using Qt6 toolkit. Some works in this direction are already done. See #75 and #77

ra3xdh, 2022-12-06:
I have recently tried to compile Qucs-S with Qt6.2.4 on Ubuntu22.04. The compilation is failing. There are remaining API incompatibility issues between Qt5 and Qt6. I have summarized found issues in the top message and added todo list. Currently QRegExp usage prevents Qucs-S from compilation with Qt6. The help with QRegExp porting is highly appreciated.

dsm, 2022-12-22:
I'll replace all foreach loop to c++11 range based for loop and code use so many NULL usage replace all to nullptr is good for future.

dsm, 2022-12-23:
clazy check help to migration to qt6 - https://invent.kde.org/sdk/clazy#list-of-checks

dsm, 2022-12-23:
setShortcut(Qt::CTRL+Qt::Key_C) is deprecated qt6

valid usage below.

setShortcut(QKeySequence(Qt::CTRL | Qt::Key_C))
QKeySequence(QKeySequence::Print);
QKeySequence(tr("Ctrl+P"));
QKeySequence(tr("Ctrl+p"));
QKeySequence(Qt::CTRL | Qt::Key_P);

QKeySequence(Qt::CTRL + Qt::Key_P); // deprecated

ra3xdh, 2022-12-23:
> setShortcut(Qt::CTRL+Qt::Key_C);// is deprecated qt6

I have this to the TODO list for this issue.

> I'll replace all foreach loop to c++11 range based for loop

Yes, the help is appreciated. It is not complex task, but will require some time. There are 281 occurrence of the `foreach` in the project. You may also look at the #194 It should be sufficient to make a simple replace for the most of QRegExp/QRegExpValidator But some complex regular expressions may require a revision. 

> code use so many NULL usage replace all to nullptr

Yes, the usage of NULL may produce warning for modern C++. But there are some parts of code in the `diagram.cpp` using the memory allocation with `malloc/free` It's better to keep this place untouched.

ra3xdh, 2023-01-15:
The deprecated `foreach` loop and `setShortcut()` were updated by the PR #206. It remains to port `QRegExp` usage.

ra3xdh, 2023-01-26:
#207 provides a fix for `QRegExp` problem and CMake options to build with both Qt5 and Qt6.

ra3xdh, 2023-01-26:
The build with both Qt5 and Qt6 is possible for the `current` branch now. Use `-DWITH_QT6=ON` cmake option to enable Qt6 build.

ra3xdh, 2023-01-26:
I have found no major issues while testing Qt6 build. Everything seems to operate correctly.

[image]

ra3xdh, 2023-01-26:
@aviadb it seems you have some Qt6 development packages missing. I have tested the Qt6 build with Ubuntu22.04 and cannot say exact package names for Fedora.

aviadb, 2023-01-26:
Ok, I got past the requirement phase after installing some qt6 devel packages.
Now, it seems to be a source related issues:
`home/aviadb/apps_src/qucs_s/qucs/components/DLS_1ton.cpp: In constructor ‘DLS_1ton::DLS_1ton()’:
/home/aviadb/apps_src/qucs_s/qucs/components/DLS_1ton.cpp:15:26: error: incomplete type ‘QObject’ used in nested name specifier
   15 |   Description = QObject::tr ("data voltage level shifter (digital to analogue) verilog device");`

This error repeats itself several time for this file:
`qucs/components/DLS_1ton.cpp`

ra3xdh, 2023-01-26:
I cannot reproduce this error using Qt6.2.4 on Ubuntu-22.04. I need to install the latest Fedora in the virtual machine and make more investigations. I have added this task to the TODO list.

ra3xdh, 2023-01-27:
@aviadb I have just added a fix for the Fedora build. Now the application could be build and run using Fedora+Qt6. Please try the latest commit on `current` branch.

ra3xdh, 2023-01-27:
> However there is an issue while dragging a component.

It's strange. I cannot reproduce this bug using the Fedora installation into the VM. There is also no bug with device dragging using physical Ubuntu installation.

dsm, 2023-01-27:
I'am not able to build with qt6.2.4 or 6.4.2 on windows 10 with mingw

dsm, 2023-01-27:
your addition for fedora has conflict for windows this will include qucs VERSION file by mistake. 

in qcompilerdetection.h 
`#ifdef __cplusplus
# if __has_include(<version>) /* remove this check once Integrity, QNX have caught up */
#  include <version>
# endif
#endif`

ra3xdh, 2023-01-27:
Yes, I can reproduce the build failure using MSYS2+Windows10. There is a naming conflict (system `<version>` include and `VERSION` file) because Windows is case-incensitive and looks for include file starting from the source root directory.

dsm, 2023-01-27:
I renamed this file to VERSION_QUCS compilation will continue but I will get this error C:/Qt/Tools/mingw1120_64/x86_64-w64-mingw32/include/rpcndr.h:64:11: error: reference to 'byte' is ambiguous

ra3xdh, 2023-01-27:
VERSION file is required by CMake to generate a proper `config.h` configuration. Renaming or moving it will break the build system. A better solution may be to exclude the source tree root from the include directories list in CMakeList.txt

dsm, 2023-01-27:
temporarily i change cmakefiles generate a proper config.h with no problem I am using mingw 11.2.0 in qt

dsm, 2023-01-31:
any progress for win10 build

ra3xdh, 2023-01-31:
Normally I dont't use physical Windows installation and MSYS2 to build Windows package. I use MXE https://mxe.cc/ cross-compile environment instead. So fixing MSYS2 build has low priority for me. It's need to check first if MXE is affected by this issue.

dsm, 2023-02-09:
I made some changes for win10 build using qt ide, some header file have use namespace std; and this will break the build windows.h file use byte conflict with std::byte. I will do some test and works I will look all includes for not used or miss used I will test windows 10 with qt 6.5.0 for futured deprecation. cmake file compiler dependent, we'll use maybe below example

```
IF(WITH_QT6)
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
ELSE()
set(CMAKE_CXX_STANDARD 11)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
ENDIF()

if (MSVC)
    # warning level 4
    add_compile_options(/W4)
else()
    # additional warnings
    add_compile_options(-Wall -Wextra -Wpedantic)
endif()
```

dsm, 2023-02-09:
qt6.5.0 win build was succesfully done but we think about version file name or handle different way. I remove below lines on simmessage.cpp and build was done. I also changed cmake for temporarily VERSION to VERSION_QUCS. any suggestion I will do and push the pr as fast as possible thanks.

```
#include <stdlib.h>
#include <iostream>
using namespace std;
```

dsm, 2023-02-13:
any suggestion for version file or send pr for above code changes ?

ra3xdh, 2023-02-13:
The #214 should resolve the problem with the version file and toplevel directory inclusion. I need to test it using MSYS2.

ra3xdh, 2023-02-13:
See https://github.com/ra3xdh/qucs_s/pull/214/commits/2c4b3c21575fe3899623b2fa120e7d7a511e9750 for solution.

dsm, 2023-02-13:
OK, I will test it with qt6.5.0 with qt ide

ra3xdh, 2023-02-13:
I have merged #214 into the `current` branch. Now you can test the Windows build with Qt6 again.

dsm, 2023-02-13:
version file not an issue anymore, I will succesfully build but some changes on simmessage.cpp I will send patch as soon as possible

dsm, 2023-02-13:
I send #215

dsm, 2023-02-13:
QT 5.14 add [QString multi-arg ](https://doc.qt.io/qt-6/qstring.html#arg-14) clazy check suggest use this example usage:

```
#include <QtGlobal>

#if QT_VERSION >= QT_VERSION_CHECK(5, 14, 0)
    s += QString(".MODEL MMOD_%1 %2MOS (%3)\n").arg(Name,mosfet_type,par_str);
#else
    s += QString(".MODEL MMOD_%1 %2MOS (%3)\n").arg(Name).arg(mosfet_type).arg(par_str);
#endif
```
[qChecksum ](https://doc.qt.io/qt-6/qbytearray.html#qChecksum) pass by pointer version deprecated used this for qt 5.9 and above we use QByteArrayView for this

dsm, 2023-02-13:
can you made a cmake modernization issue and task list. cmake 3.0 released 9 years ago, Qt build itself require v3.16 maybe we use this or above.

ra3xdh, 2023-02-13:
> QT 5.14 add [QString multi-arg ](https://doc.qt.io/qt-6/qstring.html#arg-14)clazy check suggest use this example usage

The `arg()` usage should be kept unchanged, because it's need to support Ubuntu18.04 and 20.04 that have Qt version below 5.14. The constructions of the string with `arg()` is widely used across components. If we add the conditional compilation here, we obtain a large amount of version-dependent code that is hard to maintain.  

> [qChecksum ](https://doc.qt.io/qt-6/qbytearray.html#qChecksum)pass by pointer version deprecated used this for qt 5.9 and above we use QByteArrayView for this

Yes, this call could be updated. The lowest supported version is Qt5.9.5 from Ubuntu18.04.

ra3xdh, 2023-02-13:
> can you made a cmake modernization issue and task list. 

I have just opened issue #216 You may propose tasks related to CMake here.

dsm, 2023-02-28:
export QT_DIR=/usr/lib64/cmake/Qt6 for fedora

ra3xdh, 2023-06-13:
I have added fix for the remaining deprecation warnings related to Qt6. This issue could be closed. A separate issue must be opened, if something related to Qt6 will be raised again.

closed 2023-06-13 by ra3xdh