ra3xdh/qucs_s #192: Sweep of a subcircuit parameter

0range0range, 2022-12-05:
I have a problem with sweeping parameter of a subcircuit using Ngspice,

As described here: 
https://qucs-s-help.readthedocs.io/en/latest/ASim.html?highlight=parameter%20sweep#one-and-two-parameter-sweep-controlled-simulations
it is possible to do this using ngspice specifying parameter as @dev_name[param_name]

Here is the minimal working example with a subcircuit being a voltage divider.

Main circuit:
[image]

Subcircuit:
[image]

[image]

I get the following after running simulation on a main circuit:
```
Ngspice started...
Error: no such device or model name volt_div1

```
And this error is repeated for all the parameter steps.

Am I doing something wrong and how can this be achieved?

 Also is it possible to output some values of equations from a

ra3xdh, 2022-12-05:
It is not a bug. The `@device[param]` syntax serves only for sweeping parameters of the modelcards `.MODEL`. It is not applicable for subcircuits. Furthermore it is not possible to directly sweep subcircuit parameters with Ngspice. Probably a new feature with global variable sweep may help you #189

ra3xdh, 2023-02-03:
Closing this. Let's see if the #189 will help. Otherwise the sweep of the subcircuit parameter is not possible with Ngspice.

closed 2023-02-03 by ra3xdh