ra3xdh/qucs_s #196: Simulation of custom Verilog-A modules

bertgoz, 2022-12-14:
I am a new user of Qucs-S and I have been looking for any tutorial on how to compile and simulate custom Verilog-A modules. I found a paper titled "Notes on Constructing Qucs Verilog-A Compact Device Models and Circuit Macromodels" that describes the flow in Qucs but I was not able to follow steps with Qucs-S (no qucs-core directory exist).

Could you let me know if there is info on how to use custom Verilog-A modules in Qucs-S?

Apologies if I have made any reporting mistakes. Please do not hesitate to point to me how to modify this request to make it conform to the standards.

Best Regards

ra3xdh, 2022-12-14:
Qucs-S+Ngspice doesn't support the build of Verilog-A modules on the fly. You need to install Qucsator simulation kernel and specify it in the application settings. You may be required to compile Qucdsator manually depending on your platform: https://github.com/Qucs/qucsator The procedure described in the mentioned paper "Notes on construction ..." is outdated. Follow this procedure to add Verilog-A model https://studylib.net/doc/18738014/qucs-adms-verilog-a-update---mos-ak

holvo, 2022-12-14:
With ngspice-39 (due early January 23) we will deprecate ADMS and add OSDI/OpenVAF.
OpenVAF (https://gitlab.com/DSPOM/OpenVAF) is a compiler for Verilog-A device models into a shared library, made by Semimod (https://semimod.de/).
The OSDI interface is added to ngspice (https://gitlab.com/DSPOM/ngspice). It allows dynamically linking of the shared library device models created by OpenVAF. Currently we are heavily testing this approach. Several CMC models already have been compiled.

ra3xdh, 2022-12-15:
> With ngspice-39 (due early January 23) we will deprecate ADMS and add OSDI/OpenVAF.

Hello Holger, thanks for the information. I will related features to the Qucs-S once it will be implemented in Ngspice.

ra3xdh, 2022-12-15:
> Verilog-A projects work with Qucs 0.0.19 but I was never able to get Verilog-A projects to work with the multiple versions of Qucs 0.0.20 I have.

The most probably you have MinGW version mismatch. The Qucsator and Verilog-A modules must be compiled with the same MinGW version. Otherwise you will get DLL version mismatch and application crash. The Qucs-0.0.19 is compiled with MinGW 4.8.2 But the MinGW version used for the unofficial 0.0.20 build is unknown.

ra3xdh, 2022-12-17:
> With ngspice-39 (due early January 23) we will deprecate ADMS and add OSDI/OpenVAF
Hello Holger, I have recently tried OpenVAF+Ngspice link. I was able to put into operation a simple tunnel diode model. But I didn't try more complex models like EKV. Everything seems to work correctly. So I am considering adding a support of OpenVAF into Qucs-S since v1.0.1 using this workflow https://openvaf.semimod.de/docs/getting-started/introduction/

ra3xdh, 2022-12-17:
I have added an implementation of Ngspice+OpenVAF support in the Qucs-S development plan for the next release. See #197

ra3xdh, 2022-12-18:
Closing this. You may look at the #197 to track the progress on Ngspice+OpenVAF support implementation.

closed 2022-12-18 by ra3xdh

bertgoz, 2022-12-26:
Sorry for reopening this issue. 

I have installed compiled from source qucsator, adms and qucs-s in Ubuntu 22. Building and Loading Verilog-A modules works fine (with quscator as simulator). However after the loading step when I try to simulate I get a simulator crash with a non very informative output log:

line 4: checker error, invalid definition type `Vac'
line 5: checker error, invalid definition type `TR'

ERROR: Simulator crashed!

Simulation of standard component before the Loading step works fine. Any clue of what it is going wrong?

Many thanks!

Any clues of

ra3xdh, 2022-12-26:
@bertgoz What qucsator/adms version are you using? I cannot reproduce this issue with adms v2.3.7 and qucsator-v0.0.20 from unofficial Qucs-0.0.20 beta release.

bertgoz, 2022-12-26:
qucsator 0.0.20 and adms v2.3.6 (I tried also with v2.3.7. with the same results).

The crash does not appear when I follow the Verilog-A flow with qucs 0.0.19 (haven't tried with 0.0.20 yet)

ra3xdh, 2022-12-26:
Still cannot reproduce this issue on my machine. Qucsator simulates your project without error.