ra3xdh/qucs_s #197: Add Verilog-A devices support using Ngspice+OpenVAF

ra3xdh, 2022-12-17:
Ngspice will deprecated ADMS and switch to OpenVAF compiler for Verilog-A models since version 39 https://github.com/ra3xdh/qucs_s/issues/196#issuecomment-1351948449 . It makes possible to implement Verilog-A devices support with Nsgpice+Qucs-S which is missing now.

ra3xdh, 2022-12-19:
I have added OpenVAF+Ngspice support by the recent commits on the `current` branch. The attached screenshot illustrates the simulation of the tunnel diode device with OpenVAF and Ngspice-39. 

[image]

ra3xdh, 2022-12-19:
More complex EKV model also works with OpenVAF+Ngspice-39
[image]

tomhajjar, 2022-12-21:
How will the "file" and Symbol for the Verilog-A models be done? Attached are two old Qucs Verilog-A projects. Qucs required you to create SYM and JSON files. Are you going to create a new class of File component like SpiceLibComp that loads the Verilog-A file and one can create a new Symbol?

[image]
[image]

[Qucs Verilog Projects.zip](https://github.com/ra3xdh/qucs_s/files/10279739/Qucs.Verilog.Projects.zip)

ra3xdh, 2022-12-21:
>  Are you going to create a new class of File component like SpiceLibComp that loads the Verilog-A file and one can create a new Symbol?

A new class is not required. The procedure the same as existing for Qucs. You need to open the `*.va` file in the project tree and press F9. The symbol editor opens and you can attach the `*.sym` file to `*.va` source. The `*.json'' files are created and handled automatically and user need not to create it manually. Unfortunately documentation is lacking here. The only document describing this flow could be found in this presentation from MOS-AK file:///home/vvk/Downloads/T_4_Brinson_MOS-AK_Venice_2014.pdf The procedure for Ngspice will be the same. The Qucs-S will automatically execute OpenVAF instead of ADMS.

tomhajjar, 2022-12-21:
"The symbol editor opens and you can attach the *.sym file to *.va source."

Is there an easier way to create SYM files other than cutting select sections out of existing files and then splicing them into a new file using an undefined format? You don't have a symbol editor.

[schematic_symbols.zip](https://github.com/ra3xdh/qucs_s/files/10279902/schematic_symbols.zip)

ra3xdh, 2022-12-21:
I meant a subciruit symbol editor that is called when we press F9 on subciruit or `*.va` file. The `*.sym` file from Verilog-A based project and `*.sym` file required by SpiceLibComp has the same extension but the different syntax. And there is no editor for `*.sym` files required for SpiceLibComp.

tomhajjar, 2022-12-21:
So how will the Qucs Symbol be created for the *.va file other than the way I stated above? The way it was done in https://mos-ak.org/venice_2014/publications/T_4_Brinson_MOS-AK_Venice_2014.pdf is very tedious and the average person would struggle to figure how to do it.

ra3xdh, 2022-12-21:
It's need to reuse the existing flow for the attaching symbols to the `*.va` files. Otherwise the backward compatibility with the existing projects using Verilog-A modules will be lost. The good solution is to write a better documentation than the presentation from 2014.

tomhajjar, 2022-12-21:
I went through the procedure in  https://mos-ak.org/venice_2014/publications/T_4_Brinson_MOS-AK_Venice_2014.pdf.

There is a step press F9 to "Edit Text Symbol". Can this Symbol Editor be used in other situations? It would be helpful to open other *.sym files so one could copy and paste symbols into both Verilog *.sym files and SpiceLibComp *.sch files.

[image]

[image]

[image]

ra3xdh, 2022-12-22:
> There is a step press F9 to "Edit Text Symbol". Can this Symbol Editor be used in other situations?

Yes, I am planning to reuse this symbol editor for SPICE symbols too in order to resolve #147. But I have encountered two issues:

* SPICE sym files have no proper XML header and cannot be opened by the application. This was my mistake when designing the SpiceLibComp device. The adding XML header to the existing `*.sym` files may break backward compatibility
* There is no way to insert device pins (do not confuse with ports) in the symbol mode. These pins could be only obtained automatically from subcircuit or Verilog-A. Also it will be needed to prevent user from accidentally insertions of the pins into the schematic.

After I find the solution for these issues, the editor for SPICE `*.sym` files will be implemented.

tomhajjar, 2022-12-22:
Just being able to copy and past symbol files from other sources more easily would be helpful. Trying to draw some symbols from scratch is very frustrating. Trying to make a transformer using Elliptic Arcs will drive anyone crazy. 

I found some other issues but they were using Verilog-A in Qucs 0.0.19

1. The Arrow "Head" in Symbols drawn using an Arrow is missing when placed on the schematic You must use 3 lines to make an Arrow.
3. When you choose "Load Verilog-A module" the window allows you to "Change Icon". If you choose this and then try to cancel, The JSON file associated with the Component (Rv_symbol.json) is overwritten, deleting RV.png thus corrupting the file. You are forced to recover the file from a backup if you are lucky to have one. Can you explain why Rv.png is uses a very dark background?
4. Why does "verilog-a user devices" only appear after "Load Verilog-A module"? Is there a way to create a "Compiled Library" like in QucsStudio?

[image]

[image]

[image]

[image]
[image]

tomhajjar, 2022-12-22:
1. Can you explain why Rv.png uses a very dark background? How does one know how big to make Icon.png as well as the colors and background?
2. There are multiple *.json files in each Verilog folder. Can you explain why so many? Is the format explained anywhere or how to edit them?

ra3xdh, 2022-12-22:
> Can you explain why Rv.png uses a very dark background? 

The most probably it is a transparent background. You may use any light background for you component icon. I am usually using icons with white background.

>  Can you explain why so many? Is the format explained anywhere or how to edit them?

JSON files are created automatically. User must not edit them. The format of the JSON files is not documented.

tomhajjar, 2022-12-23:
I recreated an old BSIM project that was missing it's *.va files. https://qucs.sourceforge.net/examples/BSIMTests_prj.zip

I found the *.va files and used the icons from http://qucs.github.io/qucs-manual/0.0.19/html-en/component_reference.html

The n/p noise simulations have issues. 

[Verilog-A_BSIM4_prj.zip](https://github.com/ra3xdh/qucs_s/files/10291561/Verilog-A_BSIM4_prj.zip)

ra3xdh, 2022-12-23:
> I recreated an old Verilog BSIM project that was missing it's *.va files

Yes, everything works using Qucs-S with Qucsator simulation kernel. But I will not include this example to the Qucs-S distribution. BSIM models are available for Ngspice/Xyce out of box and such example may give user a false direction. Furthermore I am planning to make deprecated Qucsator+ADMS and Qucsator netlist syntax. Qucs-S should be fully switched to the SPICE in the near future, because Qucsator development is stalled. OpenVAF+Ngspice is actively developed and seems to offer all features of Qucsator+ADMS. So Qucsator+ADMS is not needed anymore for debug and usage of Verilog-A models. Qucsator must be kept only for backward compatibility. That's why I am not inclined to add in the program distribution any new examples/documents involving Qucator and ADMS.

ra3xdh, 2022-12-23:
I have added a TODO list for this task. One issue is preventing this from inclusion in the next release. The `_props.json` files were generated by ADMS. It's need to implement a generator for these files within the Qucs-S GUI level. Now It's possible only to simulate the existing project using Verilog-A file, but not to create a new project. ADMS is required to generate `_props.json` file, but ADMS may be not installed on the system using Ngspice+OpenVAF. Furthermore ADMS should be deprecated in the next release of Qucs-S.

bertgoz, 2022-12-27:
I was testing this flow and I found a bug in the netlist syntax. In the netlist model statement there is no blank space between the Verilog-A module name and the first parameter. Simulation rans but it uses the default value for the module parameter (rather than the instance value)

Currently
>.MODEL mod_resva_resva1 resva(resistance=10.0 rmult=1 )

Should be 
>.MODEL mod_resva_resva1 resva (resistance=10.0 rmult=1 )

ra3xdh, 2022-12-27:
@bertgoz , Yes, I confirm this problem. It's strange but for my tunnel diode model Ngspice accepts both syntax (with and without space) and simulates the circuit correctly. Furthermore SPICE should accept both syntax. For example see this line from the model in the library: https://github.com/ra3xdh/qucs_s/blob/bf004c17efee5eaab8b702857cacbef400f23f82/library/SpiceOpamp.lib#L18 I think this issue should be submitted to the Ngspice-devel mailing list too with the minimal netlist to reproduce it.

ra3xdh, 2022-12-27:
Also look at this example https://sourceforge.net/p/ngspice/ngspice/ci/pre-master/tree/examples/osdi/r2_cmc/res_r2_cmc.cir There is no space between the model name and left parenthesis.

tomhajjar, 2022-12-31:
Ran into issues with the two Verilog tunnel diode models I found for Qucs.

tunn-sub.va in the Verilog examples generates errors. I found an old posting and gave up. https://github.com/Qucs/qucs/issues/318

tunnel_diode.va is in the online help and was

[image]

[image]

[image]

[Verilog-A_Tunnel_Diode_prj.zip](https://github.com/ra3xdh/qucs_s/files/10327915/Verilog-A_Tunnel_Diode_prj.zip)

ra3xdh, 2023-01-02:
> Ran into issues with two different Verilog tunnel diode models I found for Qucs.

Yes, I confirm this problem. This bug (or undocumented feature?) was presented since Qucs-0.0.19 unnoticed. The module name and the file name should be matching. Otherwise Qucs doesn't accept the icon file. Rename the Verilog-A file to `TD.va` as a workaround. I will prepare a fix for Qucs-S because OpenVAF flow also may be affected too. See updated TODO list. 

> I found an old posting that may be pertinent. https://github.com/Qucs/qucs/issues/318

The Verilog-A synthesizer mentioned in this thread was developed only for Qucs-S. Unofficial Qucs-0.0.20 release doesn't include it. You can read more here https://www.researchgate.net/publication/301319998_A_new_approach_to_compact_semiconductor_device_modelling_with_Qucs_Verilog-A_analogue_module_synthesis_VERILOG-A_ANALOGUE_MODULE_SYNTHESIS I would not recommend this Verilog-A synthesizer for generating models for examples. This feature was unmaintained because of very low demand.  It was developed in 2016 and nobody except Mike Brinson has used it since that time.

tomhajjar, 2023-01-02:
As a side note, I ended up using Windows Paint 3D to edit diode.png into tunnel_diode.png. It was far easier versus GIMP or other programs I tried. The ability to copy a pixel of the desired color and then multiply in X/Y using the mouse made it easy. 

Any reason the tunneldiode.png in Qucs is two diodes K-K like a TVS diode, versus the "normal" tunnel diode sym used in all the examples?

https://www.electronicshub.org/types-of-diodes/

[Tunnel_Diode_png.zip](https://github.com/ra3xdh/qucs_s/files/10332496/Tunnel_Diode_png.zip)

ra3xdh, 2023-01-02:
> Any reason the tunneldiode.png in Qucs is two diodes K-K like a TVS diode

Qucs doesn't include tunnel diode model in system libraries. I have never seen such symbol (two diodes) in Qucs. Mike always used a symbol like this for the tunnel diode. https://en.wikipedia.org/wiki/Tunnel_diode#/media/File:IEEE_315-1975_(1993)_8.5.7.1.c.svg The two K-K diodes symbol probably was assigned with the tunnel diode by mistake.

ra3xdh, 2023-01-02:
> As a side note, I ended up using Windows Paint 3D 

I am also using free MSPaint clone KolourPaint from KDE environment to edit device icons. Gimp is overkill for this task.

tomhajjar, 2023-01-02:
Both Qucs 0.0.19 and Qucs 0.0.20 use tunneldiode.png in the nonlinear components library. I'm sure it was the wrong symbol for a very long time. 

[image]

ra3xdh, 2023-01-02:
> Both Qucs 0.0.19 and Qucs 0.0.20 use tunneldiode.png in the nonlinear components library. I'm sure it was the wrong symbol for a very long time.

Yes, this symbol is wrong. It is Gunn diode, but not a tunnel diode.

tomhajjar, 2023-01-02:
Some would consider it a Euro version of the TVS diode. Gunn doesn't generally use the center vertical line. 

Either way, I haven't used a Tunnel or Gunn diode in decades. ;-)

[wrong_png.zip](https://github.com/ra3xdh/qucs_s/files/10332628/wrong_png.zip)

tomhajjar, 2023-01-03:
https://github.com/zergud/qucs_s/blob/current/qucs/bitmaps/tunneldiode.png

ra3xdh, 2023-01-06:
I have implemented a generator for `_props.json` files with the recent commits on the `current` branch. Now Qucs-S can generate all necessary files for Verilog-A symbol without ADMS.

tomhajjar, 2023-01-07:
I ran into issues making separate "n" and "p" components from a "unified" Verilog model, each with it's own icon. 

For example I found multiple sources of "EKV26....va". The one supplied with Qucs 0.0.19 source, EKV26MOS.va is a unified model. I couldn't figure out how to make separate n and p versions with separate icons work so I made a single model using an "polarity free" icon without an "arrow". Problem is the model defaults to "n" and no parameter selections I tried would make the "p" model work. 

I then found three "EKV26....va" models supplied with QucStudio. EKV26nMOS.va, EKV26pMOS.va and a unified model EKV26pnMOS.va. I used EKV26nMOS.va and EKV26pMOS.va to make the two n/p versions. I didn't try EKV26pnMOS.va since I couldn't figure out how to assign the same EKV26pnMOS.va file to separate n and p versions with different icons.

[Verilog-A_EKV_2v6_prj.zip](https://github.com/ra3xdh/qucs_s/files/10366510/Verilog-A_EKV_2v6_prj.zip)

ra3xdh, 2023-01-07:
> I ran into issues making separate "n" and "p" components from a "unified" Verilog model, each with it's own icon. 

It is not possible to assign more than one symbol to one Verilog-A model. Qucs supports only one one symbol for one Verilog-A module. Adding multiple symbols support will require redesign of the Verilog-A subsystem and I dont't consider implementation of this feature in the near feature because of low demand on Verilog-A modelling with Qucs-S among the Qucs-S users.

ra3xdh, 2023-01-16:
I have added a fix for module name and file name mismatch. Now Qucs-S should accept Verilog-A model if the file name and module name are different.  This fix is availale on the `current` branch.

ra3xdh, 2023-02-02:
I have upload the tunnel diode example illustrating the usage of the Verilog-A+OpenVAF

tomhajjar, 2023-02-02:
Did you fix the tunnel diode png or use the one I created?

ra3xdh, 2023-02-02:
The example for Ngspice+OpenVAF contains correct symbol for the tunnel diode. But I didn't fix the tunnel diode in the system devices. It is Qucsator-only device and not presented in Ngspice. I don't consider fixing such devices, because Qucsator is kept only for backward compatibility. Also Ngspice+OpenVAF will serve as the replacement for Qucsator+ADMS.

bertgoz, 2023-02-12:
What is the procedure to plot voltage nets/branch currents/variables that are inside the Verilog-A module?

ra3xdh, 2023-02-12:
There is no way to output the voltage/current from inside Verilog-A module using Ngspice `v()` or `i()` operation. Use debug printing with `$display` and view simulation log (F5) or make internal node external for debug time.

bertgoz, 2023-02-14:
Thanks, unfortunately $display causes openvaf (23_2_0) to not compile the module

bertgoz, 2023-02-14:
Thanks, unfortunately $display causes openvaf (23_2_0) to not compile the module

>  fatal error LNK1120: 1 unresolved externals

ra3xdh, 2023-02-14:
It's strange. The `$display` works on my OpenVAF 22.12.0

ra3xdh, 2023-02-14:
Probably you are getting this error, because some system C++ libraries are not installed. It's need to ask OpenVAF team on this.

bertgoz, 2023-02-14:
FYI https://github.com/pascalkuthe/OpenVAF/issues/39

metroid120, 2023-02-17:
we are working on the issue at OpenVAF/SemiMod, thanks for reporting

bertgoz, 2023-02-21:
@ra3xdh I was able to save both Verilog-A module internal nets and variables following the instructions at
https://sourceforge.net/p/ngspice/discussion/133842/thread/3a9dd72d33/
Any chance this could be supported in Qucs-S?

ra3xdh, 2023-02-23:
@bertgoz I didn't know about this feature of Verilog-A. I always used debug printing for Verilog-A models debugging. You may only use Nutmeg scripting simulation to make an output from these variables. This example https://qucs-s-help.readthedocs.io/en/latest/ASim.html#spectrum-analysis-with-ngspice-and-nutmeg-scripting could be used as a start point. It is totally outdated, because I have added a Spectrum analysis in the recent releases. But it gives an impression how to use the scripting simulation. 

> Any chance this could be supported in Qucs-S?

I am not planning to add an automatic recognition of the internal Verilog-A nodes in Qucs-S. To be honestly the debugging of the Verilog-A is not needed for 99% of users. I have more important task related to Verilog-A now. It's need to provide a way to include Verilog-A models into the Qucs-S libraries.

ra3xdh, 2023-06-02:
I consider the tasks related to Verilog-A support using Ngspice+OSDI are done.  I am not planning to add any new features related to Verilog-A in the near future. Closing this ticket as completed.

closed 2023-06-02 by ra3xdh