ra3xdh/qucs_s #198: Add symbols for 4 and 5-terminal SPICE BJT devices

ra3xdh, 2022-12-20:
The 5-terminal BJT device is missing in Qucs-S. The HICUM and MEXTRAN models require 5-terminal BJT symbol.

tomhajjar, 2023-02-05:
I added 5 terminal BJTs and thyristors to my symbol file. 

[image]

[schematic_symbols.zip](https://github.com/ra3xdh/qucs_s/files/10611490/schematic_symbols.zip)

ra3xdh, 2023-03-04:
I have just added unified symbols for 4 and 5 pin BJT devices. New devices will be available since v1.0.2 release.
[image]

ra3xdh, 2023-03-06:
Closing this ticket as completed

closed 2023-03-06 by ra3xdh

dwarning, 2023-11-26:
@ra3xdh and @tomhajjar: many thanks for your effort.
One remark: All latest compact models in Verilog-A like mextram, hicum, bsimcmg, psp and other get their default temperature over circuit temperature in VA code ($temperature). Device (instance) temperature can be changed by dtemp, trise or dta individual. (I belief it is a CMC rule.)
So this device symbols should be red.

tomhajjar, 2023-11-26:
Agreed. 

Changing colors requires a lot of clicks. There is no global command to alter selected line colors or widths.