ra3xdh/qucs_s #208: .STEP command with Xyce scipt element

georgtree, 2023-02-03:
Hello,

I faced with the problem - if I add command .STEP in Xyce script element, then internal graphic tool doesn't show the waveforms with different steps. With Qucs elements Parameters Sweep it works.

ra3xdh, 2023-02-03:
Yes, I confirm this. The `.STEP` inside the XYCE script doesn't work. The usual parameter sweep is not affected. I cannot provide a workaround. This feature was not tested since five years, so I doubt it was operational before.

ra3xdh, 2023-02-03:
There is no bug here, but lacking documentation on this rarely used simulation mode. Qucs-S can recognize parameter sweep automatically only if XYCE STD format is used for the output. Look at the attached simulation example. Also Qucs-S cannot extract parameter variable values and replaces it with curve numbers.
 
[image]

tvrusso, 2023-02-04:
Question:  Is this because of how Xyce outputs each step's simulation results as a separate "plot" in the rawfile output?  I could see how that might be a problem for most consumers of rawfiles.

Another question:  Would the issue of not being able to label the curves with parameter value (and having to use curve number instead) be helped by explicitly printing the parameter value on the .print line, so that the value can be used as a curve label somehow?

.STEP may be rarely used among users of Qucs-S, but it is a powerful feature that is very frequently used by Xyce users, especially with its ability to sweep parameters over values in a data table.

ra3xdh, 2023-02-04:
Hello Tom,

>  Is this because of how Xyce outputs each step's simulation results as a separate "plot" in the rawfile output?

I cannot say what was the reason not to regonize parameter sweep from SPICE-raw files. I designed the Xyce scripting simulation almost five years ago and faced some problem. Then I decided to restrict only recognition of the STD files.

>  Would the issue of not being able to label the curves with parameter value be helped by explicitly printing the parameter value on the .print line

Yes, this probably would help, but I don't consider this solution. The scripting simulation "Xyce script" has very low demand and used only for debug or research purposes. The "Parameter sweep" is recommended for normal usage. It has no problems with variable recognition. 
.
> .STEP may be rarely used among users of Qucs-S, but it is a powerful feature that is very frequently used by Xyce users

I meant .STEP usage inside scripting simulation "Xyce script" not parameter sweep analysis. I had feedback from 2 or 3 users that use it since five years. And there is no documentation how to properly use scripting mode with Qucs-S because of low demand. Normally .STEP is used with "Parameter sweep" analysis which works as expected and supports step varible recognition etc.

ra3xdh, 2023-02-13:
Assume the question is resolved. Closing this.

closed 2023-02-13 by ra3xdh