ra3xdh/qucs_s #211: Qucs Legacy diode issue

tomhajjar, 2023-02-04:
FYI

The Qucs Legacy diode has a Bv of 0 volts. It doesn't work for ngspice.

[image]

ra3xdh, 2023-02-04:
Yes, Ngspice team has recently modified the diode model. I cannot say since what version exactly. But I don't see a big issue here. The fix can wait till v1.0.2. I will not reupload the v1.0.1 packages. Diode with default model should not be normally used. This component is moved to "Qucs legacy devices" group. It's recommended to use device from library red SPICE diode device that requires to attach some non-default model. As a workaround I recommend you don't use the default model and insert rectifier diode device from library.

ra3xdh, 2023-02-04:
I have fixed this on `current` branch. The fix will be available since v1.0.2 release. Closing this.

closed 2023-02-04 by ra3xdh