ra3xdh/qucs_s #212: Thyristor testing (Triac, SCR, Diac and Sidac)

tomhajjar, 2023-02-06:
Having issues getting various Thyristor Spice models (Triac, SCR, Diac and Sidac) to work under Qucs-S. I have tested many in LTSpice and a few using ngspice. 

For example, AC_Dimmer_3.sch is tested under Qucs-S and KiCad, both using ngspice. I cannot get the Qucs-S simulation to agree with the KiCad one. 

[image]
[image]

[Diac_Triac_SCR_prj.zip](https://github.com/ra3xdh/qucs_s/files/10636802/Diac_Triac_SCR_prj.zip)
[KiCad_Triac.zip](https://github.com/ra3xdh/qucs_s/files/10636804/KiCad_Triac.zip)

ra3xdh, 2023-02-06:
The KiCAD and Qucs-S uses different Ngspice version. KiCAD uses Ngspice as the shared library (DLL or SO module), but Qucs-S uses Ngspice as the standalone application. My version of KiCAD using Ngspice-36 shared library cannot run your schematic even using "LTspice compatibility mode". It tells me `if no such function` error. I think the difference between two Ngspice versions may be explained LTspice compatibility mode implementation that was improved in the recent versions. 

Adding a thyristor devices library for Qucs-S is a good idea. I have updated TODO list. Your project containing thyristor models could be used as the starting point. But the models need to be rewritten to get rid of LTspice constructions like `if` operator which is not implemented in Ngspice. Always using LTspice compatibility mode to simulate schematics using one specific library is not a good idea.

tomhajjar, 2023-02-06:
I used ngspice 38 via DuSpice and confirmed Triac_Dimmer.cir works the same as in KiCad. No idea why it fails under Qucs-S.

Triac_Dimmer.log indicates a number of warnings and unrecognized parameters but the simulation finishes.

[image]

[image]

[KiCad_Triac.zip](https://github.com/ra3xdh/qucs_s/files/10664107/KiCad_Triac.zip)

ra3xdh, 2023-02-06:
I need the `Triac_dimmer.cir` that was used standalone run with DuSpice. I need to compare netlist this Qucs-S and netlist from KiCAD line by line.

tomhajjar, 2023-02-06:
Ooops. I sent the wrong file. 
[Thyristors.zip](https://github.com/ra3xdh/qucs_s/files/10664219/Thyristors.zip)

ra3xdh, 2023-02-06:
I made some investigations and found that it is again not a bug in Qucs-S. You have different AC source parameters and TRAN simulation settings in manually generated netlist and Qucs schematic. After applying the correction I obtained identical result. See the attached screenshot.
[image]

tomhajjar, 2023-02-07:
Fixed the simulations. The various models seem to function. Triac, SCR, Diac and Sidac. 

Sidac is a newer type of Diac used for surge protection. It can handle far more power than a Diac. Many more Sidac's are sold each year than Diac's. 

Shockley or 4-layer diodes haven't been sold in decades. Diac/Sidac are effectively two Shockley diodes in anti-parallel. 

[Diac_Triac_SCR_prj.zip](https://github.com/ra3xdh/qucs_s/files/10670755/Diac_Triac_SCR_prj.zip)

tomhajjar, 2023-10-25:
I redid the models for a DB3 Diac, L6008L6 Triac and S8020L SCR . They should do the job.

[image]
[image]
[image]

[image]

[Diac_Triac_SCR.zip](https://github.com/ra3xdh/qucs_s/files/13168063/Diac_Triac_SCR.zip)

ra3xdh, 2023-10-26:
I have uploaded `Thyristor.lib` library containing diac, SCR, and Triac devices. This library will be available since the next release. Closing ticket as completed.

closed 2023-10-26 by ra3xdh