ra3xdh/qucs_s #213: Opto-Coupler, Isolator, Triac Library

tomhajjar, 2023-02-10:
I'm working on a new Opto-Coupler, Isolator, Triac Library but I'm having problems getting Spice models that work under LTspice to work under ngspice. The first device I'm working on is a Photovoltaic Isolator or PVI. They have been around for decades with modern versions having much faster response time.  The most popular device is an Infineon PVI5050N. Unfortunately the factory model is unavailable and my attempts at creating a fully working version have failed. 

My next choices are the Vishay VOM1271 and Avago ACPL-K30T. 

VOM1271.cir created by the late LTspice contributor Jim Thompson, has been extensively tested under LTspice but I cannot get it to work under ngspice. Vishay supplied VOM1271T.cir doesn't work for LTspice or ngspice. 

Avago supplied ACPL-K30T.cir works under LTspice but give erroneous data under ngspice using LTspice compatibility mode. 

[image]
[image]
[image]
[image]

[Photo_Voltaic_Isolators_prj.zip](https://github.com/ra3xdh/qucs_s/files/10709449/Photo_Voltaic_Isolators_prj.zip)

tomhajjar, 2023-02-10:
First cut at MOC3043M Opto Triac. Works in LTspice but not ngspice. Looking at other spice models.

[image]

[Opto Triac_prj.zip](https://github.com/ra3xdh/qucs_s/files/10711582/Opto.Triac_prj.zip)

ra3xdh, 2023-02-11:
> OnSemi MOC3043M Opto Triac works in LTspice but not ngspice

It seems the LTSpice compatibility mode in Ngspice is not fully implemented. At first I corrected subcircuit to use SpiceFile device instead of SpiceLibComp. The output of the Ngspice became more verbose. It reports error in the `IF` statement in the netlists and doesn't accept `VSWITCH` model. After I set `ngbehavior=ltpsa` the schematic starts to work, but the result is wrong. I suspect this model requires to be rewritten to work with Nsgpce.

tomhajjar, 2023-02-12:
I ran into an issue with the BRT22x Opto Triac simulations. I found a typo in BRT22H.sch. It points to the model BRT22M.cir. I then renamed it to BRT22M.sch and the simulation fails. I then made a "fresh" BRT22H.sch also pointing to the model BRT22M.cir and it fails. I tried every combination of making fresh schematics and I could not get the simulations to work. I compared all models and schematics with a file comparison tool and cannot find anything that would cause issues.  Only the original files BRT22H.sch and BRT22H_TRAN.sch I sent "work". 

As a side note, the BRT22x.lib spice models have 6 pins with two unused pins having the same name, "NC". They also have two "dummy" components C1and R4 connected to NC and 0. Qucs/ngspice wouldn't work unless I depleted the two "NC" pins. I named the models BRT22x.cir. The modified models worked in LTspice.

.SUBCKT BRT22M A C NC MT2 NC MT1
X1       MT2 GATE MT1 BRT22M_tr
D1      10 C Demit
V3      11 10 0Vdc
R3       A 11  1u TC=0,0
R4	   0 NC 1MEG
C1	   0 NC 1u

[image]

[image]

[image]

[image]

[BRT22x_prj.zip](https://github.com/ra3xdh/qucs_s/files/10715070/BRT22x_prj.zip)

tomhajjar, 2023-02-14:
Vadim

Can you confirm if the issue above is a bug?

ra3xdh, 2023-02-14:
The problem with NC node is not a bug. It is a floating node that has no current and may cause a convergence issue with Ngspice. LTspice may automatically add parasitic from floating node to the ground, but Ngspice has no such feature.

Regarding the problem with path to the model file I cannot reproduce this issue. Probably there is something related to the relative search paths.

tomhajjar, 2023-02-14:
There is something wrong with Qucs-S. See the logs (renamed for clarity). 

The two netlists generated by Qucs (also renamed for clarity) work if I use DuSpice as the GUI. 

[Qucs_netlist_logs.zip](https://github.com/ra3xdh/qucs_s/files/10737155/Qucs_netlist_logs.zip)
[Qucs_netlists.zip](https://github.com/ra3xdh/qucs_s/files/10737154/Qucs_netlists.zip)
[ngspice_data_logs.zip](https://github.com/ra3xdh/qucs_s/files/10737163/ngspice_data_logs.zip)

tomhajjar, 2023-02-15:
I've given up on the OnSemi MOC3043M LTspice model.

An OnSemi MOC3082M LTspice model from the same source (https://groups.io/g/LTspice) seems to work OK. 

[MOC3082M_Opto_Triac.zip](https://github.com/ra3xdh/qucs_s/files/10738428/MOC3082M_Opto_Triac.zip)

ra3xdh, 2023-02-15:
> I've given up on the OnSemi MOC3043M LTspice model.

I have pointed above that LTSpice compatibility mode is not 100% LTSpice compatible. The most probably the problem is in default diode model that is used by MOC3043. It's unknown which parameters has the default diode and NPN devices in LTspice. By default model I mean the following usage `.MODEL D D`

tomhajjar, 2023-02-15:
First cut at an NPN output opto-isolator. 

4N25.cir is a model from the 90s and seems to work OK. More complicated than ones from LTspice.

4N35.cir from Vishay using my mods is not working. 4N35.lib is original. 

[4N25-35.zip](https://github.com/ra3xdh/qucs_s/files/10746471/4N25-35.zip)

ra3xdh, 2023-02-16:
4N35 contains model contains table-dependent source that is not supported by Ngspice. It is unclear how to modify it.

tomhajjar, 2023-02-16:
I found another Opto Isolator that uses the same model "architecture" that many in the LTspice forum use, MOC207. It also provides a Base connection. The model line "G1 3 5 N003 2 0.724m" sets the CTR and is the only difference between devices.

I also redid ALL the schematic symbols since I accidently mixed blue and black colors. 

[image]

[Opto_Isolators_prj.zip](https://github.com/ra3xdh/qucs_s/files/10758519/Opto_Isolators_prj.zip)

tomhajjar, 2023-02-22:
Finally have a working Photovoltaic Isolator (PVI) model, the Avago ACPL-K30T. Guys at ngspice came up with the fix. 

They are making changes to ngspice so these modifications ultimately won't be required. 

ACPL-K30T.cir requires ngbehavior=ltpsa. 

ACPLK30T.cir will work without ngbehavior=ltpsa. 

https://sourceforge.net/p/ngspice/discussion/120973/thread/8c874eb3ec/

[image]

[image]

[ACPL-K30T_PVI.zip](https://github.com/ra3xdh/qucs_s/files/10805431/ACPL-K30T_PVI.zip)

tomhajjar, 2023-02-22:
I redid a number of components in the Opto Isolators project. 

4N25 and MOC207 NPN Optocouplers. MOC207 also models the base connection which is desirable. Neither require ngbehavior=ltps

BRT22H/M and MOC3082 Opto-Triacs. MOC3082 does not require ngbehavior=ltps

It would be good to document that subcircuit.sch file names CANNOT be the same as the subcircuit.cir file name. The error messages created by ngspice are confusing, masking what the real problem is. 

[Opto_Isolators_prj.zip](https://github.com/ra3xdh/qucs_s/files/10806046/Opto_Isolators_prj.zip)

ra3xdh, 2023-10-24:
I have added `Optocoupler.lib` library by the recent commit. This library will be available since the next release.

ra3xdh, 2023-10-24:
TRIACs and optothyristors need to be modified to ensure compatibility with Ngspice. Every occurrence of the `IF` statement should be replaced by ternary operator.  This will allow to use the library without the need to set the compatibility options for Ngspice.

ra3xdh, 2023-10-24:
Help is appreciated for this issue and #212. The C++ knowledge is not needed to modify the models.

tomhajjar, 2023-10-25:
On 2/15/23, you modified a MOC3082 Opto Triac Spice model done by Helmut Sennewald. Is it usable to make into a library device?

[image]
[image]
[image]

[MOC3022_Opto_Triac_prj.zip](https://github.com/ra3xdh/qucs_s/files/13157285/MOC3022_Opto_Triac_prj.zip)

tomhajjar, 2023-10-25:
I dug up a very old 2N5568 Triac Spice model done by Intusoft. I haven't had a chance to test it...

[2N5568_Triac.zip](https://github.com/ra3xdh/qucs_s/files/13158344/2N5568_Triac.zip)

ra3xdh, 2023-10-25:
> On 2/15/23, you modified a MOC3082 Opto Triac Spice model done by Helmut Sennewald

Yes,  I will add this device ti the library.

> I dug up some old SCR and Triac Thyristor models that might be usable

I have checked the SPICE sources and found that these models don't contain LTspice/PSpice specific definitions. It could be accepted for Qucs-S library.

ra3xdh, 2023-10-25:
I have added MOC3082 model. The new `Optocoupler.lib` library will be available since the next version. Thyristor library will be provided in #212. Closing this ticket as completed.

closed 2023-10-25 by ra3xdh

tomhajjar, 2023-10-26:
Is the ACPLK30T Photovoltaic Isolator usable in the optocoupler library? I realize I sent confusing and cluttered messages and files for this device. The device was modified by the ngspice guys to be compatible. 

[image]
[image]

[Photo_Voltaic_Isolator_prj.zip](https://github.com/ra3xdh/qucs_s/files/13179454/Photo_Voltaic_Isolator_prj.zip)

ra3xdh, 2023-10-26:
I didn't include the ACPLK30T device. I will add it on the next library update.

tomhajjar, 2023-12-05:
Just confirming that the ACPLK30T Photo Voltaic Isolator will make it into the next release. It can be included in Optocoupler.lib if you don't want to make another library.

tomhajjar, 2024-03-16:
Confirming that the ACPL_K30T Photo Voltaic Isolator will make it into the next release. It can be included in Optocoupler.lib if you don't want to make another library.

I tested it recently in Trainline_Driver.sch. It needs a resistor load in the MegOhm range

[ACPL-K30T_PVI.zip](https://github.com/ra3xdh/qucs_s/files/14623748/ACPL-K30T_PVI.zip)