ra3xdh/qucs_s #218: Qucs-S generated log.txt does not match ngspice generated spice4qucs.log

tomhajjar, 2023-02-15:
When comparing "log" files generated by Qucs-S versus ngspice (DuSpice), I found log.txt is missing critical info contained in spice4qucs.log. It makes it difficult to diagnose what is wrong.  

[image]

[image]

[image]

[Qucs-ngspice_logs.zip](https://github.com/ra3xdh/qucs_s/files/10737958/Qucs-ngspice_logs.zip)
[ngspice_data_logs.zip](https://github.com/ra3xdh/qucs_s/files/10737946/ngspice_data_logs.zip)

ra3xdh, 2023-02-15:
There is no bug here. I have already explained you that Qucs-S doesn't generate log file by itself. It shows the log obtained from Ngspice as is. It doesn't hide any information. I launched Ngspice from the command line with your netlist and obtained exactly the same log as shown by Qucs-S. Qucs-S dosn't apply any Ngspice options to suppress the log output.  Your DuSpice installation may apply some hidden option to Ngspice that may cause the difference in logs. 

[image]

ra3xdh, 2023-02-15:
Also I have explained you before that you are using `SpiceLibComp` device wrong. It is designed for direct placement on the schematic and assign the symbol from the `*.sym` file. It may cause `nested subcircuits` error if wrapped in subschematic. Use the `SpiceFile` device instead. The `nested subcircuit` error has gone with it. 

[image]

tomhajjar, 2023-02-15:
If I'm not supposed to use SpiceLibComp to make a subcircuit, how do I specify a specific component, MOC3082 inside MOC308x.lib using SpiceFile?

[image]

[image]

[MOC3082M_TRAN.zip](https://github.com/ra3xdh/qucs_s/files/10743585/MOC3082M_TRAN.zip)

ra3xdh, 2023-02-15:
> If I'm not supposed to use SpiceLibComp to make a subcircuit, how do I specify a specific component, MOC3082 inside MOC308x.lib using SpiceFile?

Copy the models for each device into separate CIR file.

tomhajjar, 2023-02-15:
Yes obviously but I have been using SpiceLibComp this way for a long time and it generally works. Seems like a step backward.

ra3xdh, 2023-02-15:
> I have been using SpiceLibComp this way for a long time and it generally works

There is no warranty of proper operation if used inside a subcircuit. But for some cases it will work properly. Sometimes it may produce a nested subcircuit error. Also `SpiceLibComp` cannot produce a proper library entry because it still reference the library file using `.INCLUDE`. So it is not recommended if it is planned to generate the library using *Project->Create Library*

tomhajjar, 2023-02-15:
So I converted the files and no luck. ngspice does not like SpiceFile using the "new" MOC3082.cir 

The original MOC3082M.sch uses SpiceLibComp pointing to MOC3082 inside MOC308x.lib and it works

The new MOC3082.sch using SpiceFile and the new MOC3082.cir doesn't work

I created MOC3082_cir.sch using SpiceLibComp pointing to MOC3082 inside MOC3082.cir and it works

[MOC3082.zip](https://github.com/ra3xdh/qucs_s/files/10744673/MOC3082.zip)

ra3xdh, 2023-02-15:
It is again not a bug of Qucs-S or Ngspice. Your `MOC3082.cir` file is corrupt and contains unallowed unicode symbols. Fix this file first.
[image]

ra3xdh, 2023-02-15:
After fixing the corrupted symbols and renaming MOC382 to MOC3082_sub everything works fine. Otherwise Ngspice reports `Duplicate subcircuits` error.
[image]
[image]

tomhajjar, 2023-02-15:
I cannot find the corrupted symbols. I used multiple text editors and cannot see them. Only thing I see is µ versus u

ra3xdh, 2023-02-15:
> Only thing I see is µ versus u

Yes, the µ symbol is not allowed for Ngspice. It can recognize only ASCII symbols, but not unicode.

ra3xdh, 2023-02-15:
I have ported MOC3082 to Ngspice syntax, so it works without setting `ltpsa` mode now. The model is below:

~~~
* MOC3082  I_trig=10mA
*  D+  D-  MT2  MT1
*
.subckt moc3082_sub 1 2 3 4
.PARAM Itrig=10m
.PARAM RH1=20k
.PARAM RH2=20k
.PARAM RH3=16.7k
Q1 vb1p vb1 4 0 NPN1
Q2 vb1 vb1p vd1 0 PNP1
R3 vb1 4 {RH2}
D1 1 2 DL
R1 ctrl10 4 1
C1 ctrl10 4 10u
R2 ctrl1 vb1 {RH1}
R4 vd1 vb1p {RH3}
B1 ctrl10 4 I=-500*I(D1)*3m/Itrig
R7 vb2p 4 {RH3}
R6 vd2 vb2 {RH2}
Q3 vb2p vb2 vd2 0 NPN1
Q4 vb2 vb2p 4 0 PNP1
D3 vd2 3 D1
D2 3 vd1 D1
R5 N001 vb2 {RH1}
E1 vd2 N001 ctrl1 4 -1
R34 3 4 100MEG
R12 ctrl1 4 1
B2 ctrl1 4 V=(V(ctrl10,4)>1.45 && abs(V(34)) < 12?2:0)
C2 ctrl1 4 1u
B3 34 0 I=-V(3,4)/1000
R13 34 0 1k
C3 34 0 .1u
.MODEL PNP1 PNP(Is=1e-15 BF=10 Cjc=10p Cje=20p Tf=0.1u Ise=1e-12)
.MODEL NPN1 NPN(Is=1e-15 BF=10 Cjc=10p Cje=20p Tf=0.1u Ise=1e-12)
.MODEL D1 D(Is=0.1u Rs=2 Cj0=50p)
.MODEL DL D(Is=1e-20 Rs=5)
.ends moc3082
~~~

tomhajjar, 2023-02-15:
Not working for me. I'm baffled.....

[MOC3082_sub.zip](https://github.com/ra3xdh/qucs_s/files/10745525/MOC3082_sub.zip)

ra3xdh, 2023-02-15:
The `SpiceFile` device provides a wrapper for the inner subcircuit. If inner subcircuit and outer subcircuit name is the same the Ngspice will show "Duplicate subcircuits" error. Rename the `MOC3082_sub.sch` to `MOC3082.sch` and the error will disappear.

ra3xdh, 2023-02-17:
Closing because cannot reproduce the log file issue. Discussion regarding the optocoupler devices is not related to log file and should continue in #213

closed 2023-02-17 by ra3xdh