ra3xdh/qucs_s #219: AppImage

kdotom, 2023-02-15:
It appears that the latest [AppImage](https://download.opensuse.org/app/folder/196332) is version 1.0.0. 

Would you be able to provide a link to the AppImage for the [latest version](https://github.com/ra3xdh/qucs_s/releases)? (I believe 1.0.1?)

Thanks!
-K

ra3xdh, 2023-02-15:
Yes, I am aware about AppImage issues. OBS provides only Qt5.6 for building AppImage that is too old. The latest changes in the Qucs-S requires a newer Qt5 version. I need either to build AppImage manually on my host machine or find a way to specify a newer Qt5 version in the AppImage YAML config in OBS. Unfortunatelly the dedicated section of the OBS documentation is empty https://openbuildservice.org/help/manuals/obs-user-guide/cha.obs.package_formats.html#id-1.5.6.2.11 Please point me if you find how to specify the Qt version for OBS.

ra3xdh, 2023-02-18:
I failed to find any information how to specify the Qt version while generating AppImage with OBS. So, I have to build the AppImage manually on my host Kubuntu18.04 system.  AppImage could be downloaded here: https://github.com/ra3xdh/qucs_s/releases/download/1.0.1/Qucs-S-v1.0.1.x86_64.AppImage

ra3xdh, 2023-02-18:
It make sense to switch to GithubActions for AppImage generation in the future.

closed 2023-02-18 by ra3xdh