ra3xdh/qucs_s #220: Error: 'I' not found in the current scope

kdotom, 2023-02-15:
Error:
_'I' not found in the current scope._

shows up when building a Verilog A resistor. Am I missing something?

`

    'include "constants.vams"
    'include "disciplines.vams"
    
    module res (p, n);
    
	    inout p, n;
	    electrical p, n;
	    parameter real R=5k;
	    analog
		    V(p,n) <+ R * I(p,n);
    
    endmodule
`

Basic hello world module builds just fine:

`

      module hello_world;
        analog
          begin
            @(initial_step)
            $strobe("Hello World!");
          end
      endmodule

`

ra3xdh, 2023-02-15:
Close as not related to Qucs-S.

ra3xdh, 2023-02-15:
Are you sure that you are using OpenVAF compiler (not ADMS) and have proper simulation setting in the application settings? Qucsator and/or Xyce will not work with OpenVAF.

kdotom, 2023-02-15:
Found it! I used ' instead of ` in front of the include statements. Leaving a comment in case this is helpful for anyone.

ra3xdh, 2023-02-15:
Closing as resolved.

closed 2023-02-15 by ra3xdh