ra3xdh/qucs_s #221: Can't open schematic file

dwarning, 2023-02-15:
I build Qucs-S on Ubuntu 22.04 from github repository and installed under my HOME/.local, sure with write permissions.

If I try to open a schematic from examples I got a popup box with the message that this not seems a EEschema file. If I use the filemanager and make drag&drop the file is loaded and I can simulate.

BTW: Same happens with Qucs, which is installed in parallel.

ra3xdh, 2023-02-15:
> I build Qucs-S on Ubuntu 22.04 from github repository and installed under my HOME/.local

This installation mode is not recommended for Ubuntu. Use binary package instead: https://software.opensuse.org/download.html?project=home%3Ara3xdh&package=qucs-s 

> If I try to open a schematic from examples I got a popup box with the message that this not seems a EEschema file

It is not a bug. You have file extension conflict in your system. The `*.sch` extension is associated with KiCAD which is installed using system package manager. Qucs-S package doesn't make any file associations. Always use `File->Open` menu entry to open Qucs-S file, because the `*.sch` extensions is shared by many programs and may be ambiguous.

dwarning, 2023-02-15:
Thanks,
I know it is not a bug, but an issue.

closed 2023-02-15 by dwarning

ra3xdh, 2023-02-15:
The solution may be to use another extension for schematic document, so it will not conflict with other software. But the backward compatibility may be lost.