ra3xdh/qucs_s #222: Photovoltaic Relay Library

tomhajjar, 2023-02-16:
First cut at creating a Photovoltaic Relay for the Opto Isolator Library. The CPC1018N model is more sophisticated than the very simple factory Vishay VOR14642.

I modified IXYS model CPC1018N_R05.cir from the LTspice Forum and created CPC1018N_sub.cir. It isn't working so either I did something wrong or it needs to be modified. 

[image]

[image]

[Photovoltaic_Relays_prj.zip](https://github.com/ra3xdh/qucs_s/files/10751317/Photovoltaic_Relays_prj.zip)

ra3xdh, 2023-02-16:
> I modified IXYS model CPC1018N_R05.cir and created CPC1018N_sub.cir. It isn't working

The CPC1018N model works, but the testbench circuit is wrong. According the datasheet the switching speed of this device is in range 0.22-2ms, but you a using 50us pulse to test the device. 
[image]

tomhajjar, 2023-02-16:
Unfortunately there is limited Photovoltaic Relay models available and most are too simple to be useful if the relay is used in a circuit that switching speed is pertinent. However in most cases Photovoltaic Relays are used as power switches for slow devices where speed or response time is immaterial. I don't like that the Vishay VOR14642 model and others I found don't model the input diode.

The CPC1018N model includes:

.PARAM IFON=1m
.PARAM RON=0.65
.PARAM TOFF=250u
.PARAM COUT=25p
.PARAM CIO=1p

Is there a way to include some of these like Ron in the library so this model can be used for other relays? I realize what I did below caused a nested sub but it works.  

[Op_Amp_Sub_Param_Pass_prj.zip](https://github.com/ra3xdh/qucs_s/files/10756177/Op_Amp_Sub_Param_Pass_prj.zip)

tomhajjar, 2023-02-16:
I redid the schematic symbols since I accidently mixed blue and black colors.

[Photovoltaic_Relays_prj.zip](https://github.com/ra3xdh/qucs_s/files/10758742/Photovoltaic_Relays_prj.zip)

ra3xdh, 2023-09-26:
I have uploaded photovoltaic relays library in this repository containing some extra libraries. See here: https://github.com/ra3xdh/qucs-rus-complib/ Closing this as completed.

closed 2023-09-26 by ra3xdh