ra3xdh/qucs_s #223: Incorrect model type. Expected OSDI device

kdotom, 2023-02-16:
I'm wondering if I am missing a step or if this is a bug:

```
ngspice (within qucs_s) returns the error:
Ngspice started...
warning, model type mismatch in line
    nres1 vin vout mod_res_res1
Error on line 9 or its substitute:
  nres1 vin vout mod_res_res1
incorrect model type! Expected OSDI device
    Simulation interrupted due to error!
Note: No ".plot", ".print", or ".fourier" lines; no simulations run
Note: No compatibility mode selected!
Circuit: * qucs 1.0.1 /path/filename.sch
Simulation finished
```

This is from creating a Verilog A model of a resistor in qucs_s, building it with openVAF, and loading it into the schematic (a simple RC circuit).

Is there a step for loading the model beyond generating the schematic symbol and clicking "Load Verilog A module"?

Note: I used a normal resistor in this exact simulation file and it works just fine.

ra3xdh, 2023-02-17:
> Is there a step for loading the model beyond generating the schematic symbol and clicking "Load Verilog A module"?

No there is no other steps to load the OSDI device. Please note that Verilog-A simulation will not work outside projects. The `*.va` and `*.osdi` should be in the project directory. Otherwise Qucs will not see the module. Also please submit you schematic and VA file so I can debug this issue. Probably you are doing something wrong with schematic.

metroid120, 2023-02-17:
Just a quick comment from my side:
- You can specify the location of OSDI files with the "pre_osdi" command in the netlist
- One can configure Ngspice so that it looks for OSDI files in a specific library
This is explained here:
https://openvaf.semimod.de/docs/getting-started/usage/

kdotom, 2023-02-17:
I just reproduced the error with a basic ngspice RC sim, so I don't think the issue is with qucs.

```
*SPICE circuit

.model Nres1 res R=10k

V1 vin GND PULSE(0 5 1m 1p 1p 1m 6m)
C1 vout GND 0.1u
Nres1 vin vout
*R1 vin vout 10k

.tran 0.01m 6m
.control
pre_osdi ~/.qucs/sim_RC/res.osdi
run
set color0 = white
set color1 = black
plot vin vout
.endc
.end
```

using res.va (below / similar to [this one](https://help.simetrix.co.uk/8.0/simetrix/mergedProjects/verilog_a_reference/topics/veriloga_writingverilog_acode_aresistor.htm))

```
`include "disciplines.vams"
`include "constants.vams"

module res (p, n);
        inout p, n;
        parameter real R=5k;
        electrical p, n;

        analog
                V(p,n) <+ R * I(p,n);

endmodule
```

using just ngspice, the same error appeared

kdotom, 2023-02-18:
Got it working. Changed the module name to a different name than the osdi file. openvaf built the original file properly, so maybe ngspice just doesn't process identical names like that correctly. Thanks!

ra3xdh, 2023-02-18:
> Changed the module name to a different name than the osdi file (/ and the va file)

No, I have tried the module and file with the same name and everything works fine. This module also works on my machine as expected. I guess here is something version-dependent. Please consider submitting the issue report to OpenVAF repository.

ra3xdh, 2023-02-23:
Closing as not related to Qucs-S itself. This may be OpenVAF issue.

closed 2023-02-23 by ra3xdh