ra3xdh/qucs_s #226: uic option for .tran simulation

cznwhale, 2023-02-20:
Please add the possibility to use initial conditions in a transient analysis.
For this, it is required to add an option in the TRAN simulations block that would put "uic" text in the .tran command

ra3xdh, 2023-02-20:
Everything is already implemented. The *Initial DC=yes* option adds the UIC condition to the transient analysis. It could be combined with `.IC` virtual device. 
[image]
[image]

cznwhale, 2023-02-20:
Hi,
Thank you for the fast response. I confirm that putting initialDC=no will put "uic" on the .tran line.

pavel-kirienko, 2023-04-04:
I also stumbled upon this. Would it not be better to reflect this option in the docs in a more explicit manner?

ra3xdh, 2023-04-05:
@pavel-kirienko Currently we don't have a person assigned specially for the documentation. I am writing a new tutorial for Qucs-S but the progress is slow. Don't expect a fast update here. I can update the `InitialDC` option description in the simulation device properties to specify explicitly that is mapped to the `UIC` option.

ra3xdh, 2023-04-05:
I have added a TODO list for this. The hint in the simulation description would be helpful.

ra3xdh, 2023-04-08:
I have fixed the `InitialDC` option description to mention UIC explicitly. I cannot change the option name because it may break schematic load and Qucsator compatibility. 
[image]

closed 2023-04-08 by ra3xdh