ra3xdh/qucs_s #227: Schematic glitches fedora 37 [Qt6]

aviadb, 2023-02-20:
In the schematic viewer,
moving parts or wires makes the wires to disappear until the component is released.

 
[image]

Fedora 37, Qt6, KDE desktop with dark theme

ra3xdh, 2023-02-20:
It was already reported here https://github.com/ra3xdh/qucs_s/issues/190#issuecomment-1406371905 I remember about this issue, but unfortunately I cannot reproduce and debug it. I have tried Fedora installation into the VM and everything works as expected here. Physical installation of the Ubuntu22.04 with Qt6 also works as expected. I only may suspect this may be related to Wayland.

cznwhale, 2023-02-20:
The same behavior is on the raspberry pi version. (I use the September 22nd 2022 release, 32 bit version, Kernel: Linux 5.15.84-v7l+, on a 4B model)

ra3xdh, 2023-02-21:
The issue also may be fixed by https://github.com/ra3xdh/qucs_s/pull/214/commits/fe135fa8d45d649118ad6ce4934f781c6d8b9b55 Did you try compile the latest `current` branch?

cznwhale, 2023-02-21:
in my case (Raspberry Pi version) I used the precompiled binary (qucs-s_1.0.1-1_armhf.deb)

cznwhale, 2023-02-21:
I now realize that the same behavior is for Windows portable version (qucs_s_win64_1_0_1.zip)

ra3xdh, 2023-02-21:
I have just tested the Windows v1.0.1 package using the physical Windows installation (not VM) and also cannot reproduce the issue. Component dragging works as expected on my hardware. Therefore I cannot provide a fix, because I still cannot reproduce this bug.

aviadb, 2023-02-21:
Tried with the latest, problem still exists.
Any log dump that I can provide to help you in analyzing this behavior?

ra3xdh, 2023-02-22:
The information related to this problem is not logged by application. I suspect this may be hardware-dependent bug in Qt itself. Qt relies on OpenGL for the rendering and this may interact with the Wayland or graphic card driver in a weird way. The main problem is that I have tried different hardware including Nvidia, Radeon, and Intel and everything works as expected on my hardware. No issue observed with X11 and Windows. I still cannot reproduce this issue. And therefore I cannot provide a workaround. Contribution is highly welcome here. Currntly I can only speculate about its reason.  Also there may be something in the Qt configuration related to the rendering settings or scaling.

ra3xdh, 2023-02-22:
@aviadb You may try the fix proposed in #229 I didn't merge it into the `current` branch, you need to apply patch manually. I suspect these two issues may be related. I also cannot reproduce #229 on my hardware.

aviadb, 2023-02-22:
Hi @ra3xdh , thank you for your suggestion.

Tried to comment out this line:
[image]

No success, the issue is still there.

dsm, 2023-02-28:
I tried fedora 37 with qt6 same issue is still there.

ra3xdh, 2023-03-01:
@dsm please consider providing a fix if you can reproduce this issue. I have tried different hardware and OS combinations including WIndows10 and 11 and still cannot reproduce it. The start point for debugger may be `Component::paintScheme()` method in `component.cpp`.

dsm, 2023-03-01:
I tried something but I cannot resolve symbols redraw every move but lines are not. maybe new lines coordinate not calculate while moving.

dsm, 2023-03-03:
windows 10 with qucs 1.0.1 and current branch with qt6 have the same issue 

[image]

dsm, 2023-03-03:
if you move components using keyboard it's working. I didn't find any solution

[image]

ra3xdh, 2023-03-03:
> if you move components using keyboard it's working.

The moving of the component with keyboard doesn't invoke the `Component::paintScheme()` I guess this issue lies somewhere in in this method. I cannot provide you a more info because I cannot reproduce it on my hardware. 

> windows 10 with qucs 1.0.1 and current branch with qt6 have the same issue

It is not bug because of Qt5 migration and not related to this issue. I have just checked old Qt4 version and it also doesn't render wires while moving with mouse. But the component outline is rendered normally like on your screenshot.

dsm, 2023-03-03:
qt5 or qt6 build have same behaviour on my machine. I don't know it's qt related bug but if you not release mouse button while moving not update the wires maybe it's mouse action problem I cannot debug correctly. it's not important it's just an cosmetic bug. we will fix anytime

ra3xdh, 2023-09-26:
I still cannot reproduce this issue and provide debugging. @aviadb Could you try this patch for `schematic.cpp` and tell the result? This may fix the contour drawing.
~~~
diff --git a/qucs/schematic.cpp b/qucs/schematic.cpp
index 01fe40aa..41c883a1 100644
--- a/qucs/schematic.cpp
+++ b/qucs/schematic.cpp
@@ -462,7 +462,10 @@ void Schematic::drawContents(QPainter *p, int, int, int, int)
   {
     // QPainter painter2(viewport()); for if(p.PaintOnViewport)
     QPen pen(Qt::black);
-    Painter.Painter->setPen(Qt::black);
+    QPen pen1(Qt::black);
+    pen1.setWidth(1);
+    Painter.Painter->setPen(pen1);
+    //Painter.Painter->setPen(Qt::black);
     switch(p.pe)
     {
       case _NotRop:
~~~

aviadb, 2023-11-22:
@ra3xdh ,
The issue still persist with the suggested fix.
Using the arrow keys, I do not see the problem. Only when dragging elements with mouse.

ra3xdh, 2024-01-18:
#467 provides a fix for this issue. @aviadb , Could you test it, because I still cannot reproduce a bug with moving.

closed 2024-01-19 by ra3xdh

aviadb, 2024-01-20:
Thanks @ra3xdh, the provided solution fixes the problem.

PR #467 (2024-01-19): Fix component moving
fixes #227