ra3xdh/qucs_s #228: Transformer Library Testing

tomhajjar, 2023-02-20:
I cannot figure out how to use some of the transformer models in Transformers.lib and how to use properly use Cores.lib. 

First issue is the model XSPICEPS1S2 in Transformers.lib. On the Help web page below, it shows two parameters L and A that don't exist in XSPICEPS1S2 in Transformers.lib. The parameters ntp, nts1 and nts2 are cannot be known without knowing L and A. Another strange issue is R5 shown on the Help web page is 2k. No one would ever use a resistor on the primary to being with never mind 2k. 
When I try to use XSPICEPS1S2, it is in saturation regardless of ntp, nts1 and nts2. Obviously I could put 2k for R5 but that is not correct. 

Lastly shouldn't the XSPICE transformer models allow one to choose the core material in Cores.lib?

https://qucs-s-help.readthedocs.io/en/latest/SPICEComp.html#two-winding-transformer-model-with-in-phase-primary-and-secondary-voltages-and-winding-resistance

[image]

[Transformer_testing.zip](https://github.com/ra3xdh/qucs_s/files/10787486/Transformer_testing.zip)

tomhajjar, 2023-02-20:
I looked at Transformers.lib and it seems L and A are fixed for XSPICEPS1S2. This makes the model almost useless. These limits should be mentioned in the Library "Show model" 

.model Steel core length = 0.2 area=2e-4

[image]

ra3xdh, 2023-02-21:
> I looked at Transformers.lib and it seems L, A and core laminate are fixed for XSPICEPS1S2

Yes, this device should be corrected. The L and A parameters are indeed fixed. 

> Also, shouldn't the XSPICE transformer models allow one to choose the core materials mentioned in Cores.lib? How does one use Cores.lib

It is not possible to specify the model name as the subcircuit parameter. The only way to attach core is to create your own subcircuit as shown in the Qucs-S manual.

tomhajjar, 2023-02-21:
Can the *.sch files used to create the Help file page be released?

https://qucs-s-help.readthedocs.io/en/latest/SPICEComp.html#two-winding-transformer-model-with-in-phase-primary-and-secondary-voltages-and-winding-resistance

ra3xdh, 2023-02-21:
I don't own these schematics. The help was written by Mike Brinson around 2016. I am not sure if some schematics are remaining after seven years passed.

tomhajjar, 2023-02-21:
Mike is going to look. ;-)

tomhajjar, 2023-02-23:
I'm finding multiple issues with this library. 

Cosmetic issues:

For ease of drawing schematics or substituting one transformer with another, similar transformer types should use the symbol and pin locations. 

The five, two winding transformers use different symbols and pin locations. . 

The two, three winding (center tapped) transformers use different symbols and one pin in each was placed in the wrong location. 

Functional issues:

 XSPICEPS1S2 output polarity is flipped as shown below. As stated above, parameters N and L are missing and the user should know what the core type is. A lot more work needs to be done to confirm the functionality of the model

I'm trying to test the other models for functionality. Waiting for Mike to send the original files.

[image]

tomhajjar, 2023-02-23:
I made a first cut at new transformer symbols. They are fully symmetrical with pins on the grid. You can substitute a different model, mirror or rotate devices without having to fix the schematic. 

[image]

[symbols_transformers.zip](https://github.com/ra3xdh/qucs_s/files/10816243/symbols_transformers.zip)

ra3xdh, 2023-02-23:
The symbols in the library is not subject of change. The library was fully operational at least until Ngspice-27. And there may be users that designed circuits with old symbols. I don't know how many users used these devices. If we change the symbols in the library, the backward compatibility will be lost and **ALL** existing schematics will require redesign. This may cause a tons of bugreports why the symbols were changed.

tomhajjar, 2023-02-23:
Understand. People can use the symbols to make new transformer sub circuits.

ra3xdh, 2023-02-23:
I have attached a TODO list for this issue.

ra3xdh, 2023-06-22:
I have fixed XSPICE transformers in the library. You can get the latest library from here: https://raw.githubusercontent.com/ra3xdh/qucs_s/current/library/Transformers.lib Closing this as resolved. The solution for #231 will be added separately.

closed 2023-06-22 by ra3xdh

tomhajjar, 2023-06-22:
It will take a while to confirm all the changes but XSPICEPS1S2 still has an inverted output in the "new" Transformers.lib like it was in the old.