ra3xdh/qucs_s #229: The trail behind the model

SCP1715, 2023-02-22:
I noticed that when you move the model around the workspace, it leaves a trail behind it. I found that this is caused by the model being drawn twice when moving
I was able to fix this by commenting out the first method call:

`selElem->paintScheme(Doc)`
in 
`void MouseActions::MMoveElement(Schematic *Doc, QMouseEvent *Event)`

ra3xdh, 2023-02-22:
It's strange but I also cannot reproduce this issue on my hardware. The device moving works as expected. What is your hardware, OS and Qt version?

SCP1715, 2023-02-22:
I noticed this on windows 7, Qt 5.14 and 5.8 
[image]
[image]
This is noticeable when you move the models very slowly or very quickly

ra3xdh, 2023-02-22:
Yes, I understood how to reproduce it now. It appears only if the device is moved very slowly.

ra3xdh, 2023-09-26:
The fixing of this issue may require the redesign the schematic rendering widget that is not considered in the near future. This issue is rare to appear during the normal operation and doesn't affect the application operation. Closing as won't fix.

closed 2023-09-26 by ra3xdh