ra3xdh/qucs_s #231: Mike Brinson transformer library model test schematic fails simulation

tomhajjar, 2023-02-23:
The Full_wave.sch schematic in the TranLibTest.qucs archive done by Mike Brinson has something wrong with it. All the other schematics seem OK. I made a copy of the file and data and renamed it to "Full_wave_orig" to preserve it. 

When you open Full_wave.sch, the original test data  .

[image]

[image]

[Test_prj.zip](https://github.com/ra3xdh/qucs_s/files/10809461/Test_prj.zip)

ra3xdh, 2023-02-23:
This schematic works as expected using Ngspice-27. But doesn't produce a correct result using the latest Nsgpice-39. I suspect something has been changed in Ngspice or XSPICE since the library release. The `Transformers` library need to be fixed to provide compatibility with the latest Ngspice.

tomhajjar, 2023-03-03:
I'm trying to identify all the issues with the XSPICEPS1S2 transformer in the library. As stated before, the model lacks the ability to change L or A and it's output polarity is inverted. So I decided to recreate the XSPICEPS1S2 the subcircuit as shown in both the online help and Mike Brinson's 2016 paper. 

I can get XSPICEPS1S2 from the library to simulate as shown in XSPICEPS1S2_lib_test.sch. 

I cannot get the subcircuit I made XSPICEPS1S2_sub.sch to simulate in XSPICEPS1S2_sub_test.sch.

As a side note, the Library view of the Cores library has issues. A spurious graphic **{-}** is displayed. 

[image]

[image]

[image]

[Transformer_sub_Test_prj.zip](https://github.com/ra3xdh/qucs_s/files/10885586/Transformer_sub_Test_prj.zip)

ra3xdh, 2023-03-04:
I made some investigations on the full-wave rectifier circuit using XSPICE transformer. It seems there is something wrong inside XSPICE magnetic core model itself. It may operate correctly only if sinusoidal voltage applied to it. The summary is the following:

* The Core model works only using Ngspice-27 and only using some specific diode models connected to secondary winding. For example it fails when I replace diodes by 1N400x device from library. 
* The full-wave rectifier simulation always fails to converge when using Ngspice-39

> Maybe you can spot an obvious mistake.

The first line of ICOUPLE devices should be model name. It it SIcouple on attached screenshot. 
[image]

tomhajjar, 2023-03-05:
I cannot see an obvious mistake.

I found an old KiCad (ngspice) netlist using core and lcouple, Trans_Xspice.cir. 

I modified it and used DuSpice/ngspice.  Analysis shows the output **inverted** (probably caused by the output pins 3/4 flipped. I cannot vouch for the accuracy of the data. The core L and A values (area = 1 length =0.01) are not realistic and swapping them causes output data to be weird 

I entered the circuit Trans_Xspice.cir into Qucs-S, 1P1S_XSpice_tran_test.sch. Unfortunately I'm having issues using the XSPICE "core" Component. I cannot figure out how to enter the core data from Trans_Xspice.cir into the XSPICE core block. Every variation I tried cause ngspice errors so I ended up using the Cores Library model "SheetSteel". Again the data is **inverted** and I cannot vouch for the accuracy. 

 versus using a core model from the core Library. I cannot figure out what for mat to use for  

[image]

[image]

[image]

[image]

[Trans_Xspice.zip](https://github.com/ra3xdh/qucs_s/files/10893033/Trans_Xspice.zip)

tomhajjar, 2023-03-06:
I created 3 netlists for use with DuSpice/ngspice to figure out what's going on. I believe Qucs-S is formatting the netlist entry for the generic Core model from Components wrong.

Trans_Xspice.cir is the test netlist I found on  the web. I modified it to plot only in and out. It simulates without error. Data accuracy unknown. 

The following two netlists are generated by 1P1S_XSpice_tran_test.sch.

Trans_1p1s_lib_netlist.cir, uses the SheetSteel Core model from the Core Library

Trans_1p1s_comp_netlist.cir uses the generic Core model from Components, Core. 

Trans_1p1s_lib_netlist.cir - Simulates without error. Data accuracy unknown. 

Trans_1p1s_comp_netlist.cir - Fails to simulate. 

The "ACORE" entry in Trans_1p1s_comp_netlist.cir is missing multiple items like .model ACORE2 core ( ........). Also missing are mode=1 but I believe that is the default for the PWL core.

[Qucs_netlists.zip](https://github.com/ra3xdh/qucs_s/files/10894055/Qucs_netlists.zip)

tomhajjar, 2023-03-07:
A minor issue is the name and prefix used for the "lcouple". 

In Components -> XSPICES devices, the lcouple is named "Capital eye"couple and the prefix is "_I_"COUPLE.
__

tomhajjar, 2023-03-14:
Until the XSICE devices Component's Core is fixed I cannot recreate Mike Brinson's 2015 data. The netlist entries for "Core" are missing multiple entries.

[image]

ra3xdh, 2023-03-16:
> The "Core" netlist parameters and data structure is wrong

Ngspice should accept both modelcard card syntax. Both space separated parameters and using parentheses should be allowed. Something may have changed here between Ngspice-27 and the latest Nsgpice.

tomhajjar, 2023-03-16:
The is a lot of differences between the KiCad/ngspice manual example and what Qucs-S generated. 

Qucs-S is trying to combine the two separate "a" and "core" netlist entries into a single entry. How is this proper?

********************************************************************************************************************************
KiCad netlist Trans_Xspice.cir:

a3 (interNode_1 interNode_2) u1_iron_core
.model u1_iron_core core (H_array = [-1000 -500 -375 -250 -188 -125 -63 0 63 125 188 250 375 500 1000] B_array = [-3.13e-3 -2.63e-3 -2.33e-3 -1.93e-3 -1.5e-3 -6.25e-4 -2.5e-4 0 2.5e-4 6.25e-4 1.5e-3 1.93e-3 2.33e-3 2.63e-3 3.13e-3] area = 0.01 length = 0.01)

********************************************************************************************************************************
Qucs-S generated netlist for Trans_1p1s_comp_netlist.cir:

ACORE2 n1 n2  Length = 1 Area = 0.01
+ H_array = [-1000 -500 -375 -250 -188 -125 -63 0 63 125 188 250 375 500 1000]
+ B_array = [-3.13e-3 -2.63e-3 -2.33e-3 -1.93e-3 -1.5e-3 -6.25e-4 -2.5e-4 0 2.5e-4 6.25e-4 1.5e-3 1.93e-3 2.33e-3 2.63e-3 3.13e-3]

[Qucs_XSpice_Trans_netlists.zip](https://github.com/ra3xdh/qucs_s/files/10992520/Qucs_XSpice_Trans_netlists.zip)

tomhajjar, 2023-03-21:
Last comment:

Only the Core parameters for H. B, A and L are different for 1P1S_XSpice_test.sch and 1P1S_XSpice_Core.sch. The netlist structures should be the same but 1P1S_XSpice_test.cir and 1P1S_XSpice_Core.cir look NOTHING alike.  

1P1S_XSpice_test.cir uses ".SUBCKT Cores_SheetSteel"

1P1S_XSpice_Core.cir is missing the .SUBCKT entry

[1P1S_XSpice.zip](https://github.com/ra3xdh/qucs_s/files/11033626/1P1S_XSpice.zip)

tomhajjar, 2023-03-23:
Until Components...XSPICE...core device is fixed, if you need a Core different than in Cores.lib or used in Transformers.lib the only choice is to use a Text Editor and alter the B, H, L and A parameters manually in the .sch file.

XSPICEPS1 and XSPICEPS1S2 need A and L added. Info on Core material would be useful.

XSPICEPS1S2 output is inverted so output nodes need to be swapped. Info on Core material would be useful.

Strangely XSPICEPS1, XSPICEPS1S2, LossyTransformer1 or LossyTransformer2 don't use any of the B/H parameters in Libraries...Cores. I'm unsure where any of the B/H data in Transformer.lib and Core.lib came from. All the more reason to get XSPICE...core working. 

I tested using lcouple and Sheet Steel Core to make a non-linear inductor. AFAIK it works. 

[image]

[image]
 
[Qucs-S.zip](https://github.com/ra3xdh/qucs_s/files/11046000/Qucs-S.zip)

ra3xdh, 2023-06-22:
I have tested the provided schematic again. The problem is that you are using the system CORE device wrong. There is not bugs in netlist generator. The attached example shows how to define core model using this device. 
[image]

ra3xdh, 2023-06-22:
Here is the schematic file. 

[Core.zip](https://github.com/ra3xdh/qucs_s/files/11830235/Core.zip)

tomhajjar, 2023-06-23:
I confirmed that the changes to Transformers.lib work however XSPICEPS1S2 still has an inverted output. See my comments in #228

ra3xdh, 2023-06-23:
I have fixed polarity issue of the XSPICEPS1PS2 device. I have also found modelcard issue that may cause incorrect simulation. See the latest version here: https://raw.githubusercontent.com/ra3xdh/qucs_s/current/library/Transformers.lib

tomhajjar, 2023-06-23:
I recreated all the Mike Brinson transformer models in the online help. This folder may be useful for someone.

[Transformer_sub_Test_prj.zip](https://github.com/ra3xdh/qucs_s/files/11848239/Transformer_sub_Test_prj.zip)

closed 2023-06-23 by ra3xdh