ra3xdh/qucs_s #232: Odd Op-Amp behavior (and crash on XYCE)

auxym, 2023-02-23:
Hi, first of all thank you for maintaining qucs-s, it's much nicer to use than LTSpice, plus it runs on Linux :). Second, sorry if this all user error, I am somewhat new to electronics simulation.

I am using the following software versions:

- OS: Manjaro linux
- qucs-s
- ngspice 38-1
- `xyce-serial 7.5.0-1` installed from the AUR

Here is a simple model which attempts to model op-amp behavior in an open-loop/comparator configuration. It is fed a sine wave as input, and the output is expected to be a (nearly) square wave that goes to either supply rail.

[test1.txt](https://github.com/ra3xdh/qucs_s/files/10809640/test1.txt) (renamed .sch to .txt)

I note that qucs-s provides op-amp models under two categories: "OpAmps" and "SpiceOpamps".

**Working**:

When using an op-amp model (ad822) from "SpiceOpamps", with the ngspice solver, I appear to be getting the correct/expected result (square wave output).

**Not working:**

When using an op-amp model from "OpAmps" (tl071) with the ngspice solver, the output seems to go to one of the supply rails but doesn't oscillate with the input.

When using an op-amp model from "OpAmps"  (tl071) with the XYCE solver, the output seems to go to one of the supply rails but doesn't oscillate with the input (same as noted above with ngspice). Additionally, XYCE outputs the following warnings:

```
***** Reading and parsing netlist...
Netlist warning in file /home/francis/.qucs/spice4qucs/spice4qucs.tran.cir at
 or near line 36
 No model parameter N found for model JMOD_J1 of type PJF, parameter ignored.
Netlist warning in file /home/francis/.qucs/spice4qucs/spice4qucs.tran.cir at
 or near line 36
 No model parameter M found for model JMOD_J1 of type PJF, parameter ignored.
Netlist warning in file /home/francis/.qucs/spice4qucs/spice4qucs.tran.cir at
 or near line 38
 No model parameter N found for model JMOD_J2 of type PJF, parameter ignored.
Netlist warning in file /home/francis/.qucs/spice4qucs/spice4qucs.tran.cir at
 or near line 38
 No model parameter M found for model JMOD_J2 of type PJF, parameter ignored.
```

When using an op-amp model from "SpiceOpamps" (AD822) with the XYCE solver, XYCE crashes with the following error:

```
Xyce (serial)  started...

*****
***** Welcome to the Xyce(TM) Parallel Electronic Simulator
*****
***** This is version Xyce Release 7.5-opensource
***** Date: Wed Feb 22 20:55:36 EST 2023

***** Executing netlist /home/francis/.qucs/spice4qucs/spice4qucs.tran.cir

***** Reading and parsing netlist...
/usr/include/c++/12.2.1/bits/stl_vector.h:1123: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = Teuchos::RCP<astNode<std::complex<double> > >; _Alloc = std::allocator<Teuchos::RCP<astNode<std::complex<double> > > >; reference = Teuchos::RCP<astNode<std::complex<double> > >&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Simulation finished
```

ra3xdh, 2023-02-23:
> When using an op-amp model (ad822) from "SpiceOpamps", with the ngspice solver, I appear to be getting the correct/expected result 
> When using an op-amp model from "OpAmps" (tl071) with the ngspice solver, the output seems to go to one of the supply rails but doesn't oscillate with the input

The `OpAmps` library contains model designed for Qucsator simulation kernel. These models may be not compatible with SPICE and will provide poor simulation result. The `OpAmps` library is not recommended to use with Ngspice and kept only for backward compatibility. Use `SpiceOpamp` instead. It will be needed to blacklist the `OpAmps` library to prevent user from accidentally insertion device from it. 

> When using an op-amp model from "SpiceOpamps" (AD822) with the XYCE solver, XYCE crashes with the following error

Yes, I confirm this issue. It is related to Xyce itself and should be reported to the Xyce team. I suspect something may be wrong with `POLY` sources implementation in Xyce.

ra3xdh, 2023-02-23:
OpAmps library will be blacklisted since the next release and kept only for backward compatibility. Use SpiceOpamp.lib instead. Closing this.

closed 2023-02-23 by ra3xdh

auxym, 2023-02-24:
Thanks for looking into this and blacklisting the legacy opamps, it should make things clearer for new users.

I wanted to note that I tried most op-amp models under SpiceOpamp with ngspice, and they all seem to work as expected, except the **tl071**, which seems to stick to a power rail. Maybe there is an issue with this model?

On XYCE side, I again tried most SpiceOpAmp models, and they either crash or produce incorrect (stuck to power rail) result. I will look into reporting this to the XYCE project.