ra3xdh/qucs_s #233: Add s2p files support for Qucs-S

ra3xdh, 2023-02-24:
Ngspice supports S-parameters simulation since release 37, but the simulation of the `*.s2p` files is not supported.  There exists `s2spice` utility https://github.com/transmitterdan/s2spice designed by @transmitterdan that converts s2p files to SPICE netlists having LTSpice compatible syntax. The similar approach could be used for Qucs-S, but there exists a problem. Ngspice doesn't support the frequency dependent E-source that is used by `s2spice` to represent the `s2p` files. It's need to find an Ngspice replacement for such source type. Then the `s2p` files support could be implemented for Qucs-S.  

s2spice utiliy uses the following E-source syntax:
~~~
 E0101 101 102 FREQ {V(100,3)}= DB
+(  5.000000e+05Hz,  1.696474e-03, -1.886862e+00)
+(  1.000000e+06Hz,  3.036754e-03, -3.763558e+00)
+(  1.500000e+06Hz,  4.436143e-03, -5.651555e+00)
+(  2.000000e+06Hz,  1.758355e-03, -7.543694e+00)
+(  2.500000e+06Hz,  3.780545e-03, -9.429202e+00)
+(  3.000000e+06Hz,  2.498363e-03, -1.132884e+01)
+(  3.500000e+06Hz,  2.890581e-03, -1.325903e+01)
~~~

tomhajjar, 2023-02-25:
I'm hoping to get help from the ngspice guys on this issue. The E source format used in the s-parameter conversion file looks similar for LTspice and ngspice. Maybe the fix won't be too difficult.

LTspice
E11 11 12 FREQ {V(10,3)} = DB (1.000000e+006,-2.162013e+000,-1.673312e+001).....

ngspice
Syntax for data entry from table:
Exxx n1 n2 TABLE { expression } = (x0,y0) (x1,y1) (x2,y2)
Example (simple comparator):
ECMP 11 0 TABLE {V(10 ,9)} = (-5mV , 0V) (5mV , 5V)
An ’=’ sign may follow the keyword TABLE.

[S-parameter_files_ngspice_prj.zip](https://github.com/ra3xdh/qucs_s/files/10831679/S-parameter_files_ngspice_prj.zip)

tomhajjar, 2023-02-25:
Here is a discussion on LTspice forum about the format for "E" as used in the converter.

[E11 11 12 FREQ {V(10,3)} = DB (1.000000e+006,-2.162013e+000,-1.673312e+001)..... ](https://groups.io/g/LTspice/topic/77622721?p=Created,,,20,2,0,0::recentpostdate%2Fsticky,,,20,2,0,77622721)

tomhajjar, 2023-02-26:
Article with QuickBasic program on how to convert S-parameter files for PSpice.

[CreateS-ParameterSUBCKTinPSpice.zip](https://github.com/ra3xdh/qucs_s/files/10832840/CreateS-ParameterSUBCKTinPSpice.zip)

transmitterdan, 2023-02-26:
I'll take a look and see if there is a simple way to target ngspice.  Will take time as I am traveling a lot these days.

tomhajjar, 2023-03-02:
Dan

Thank you for looking into this.

Would it be easier to find a solution if only MAG was used instead of MAG and DB?

The original program from decades ago, S2P2LIB1.bas does a unwrap-phase but S2spice doesn't. Can this cause phase ambiguity issues?

I tested ATF10236.s2p with S2P2LIB1_fixed.bas and S2spice. You can compare the differences between the two lib files.

[S2P2LIB1_QB64.zip](https://github.com/ra3xdh/qucs_s/files/10867716/S2P2LIB1_QB64.zip)
[S2spice.zip](https://github.com/ra3xdh/qucs_s/files/10867719/S2spice.zip)

tomhajjar, 2023-03-02:
To help in testing, I created a number of test files using Qucs. 

I used Qucs since it has the ability to convert a .dat file (with S-parameter data) to a .s2p file using Qucsconv,

Files:
137MHz_BPF.sch, 137 MHz BPF using L and C

137MHz_BPF.s2p, created from 137MHz_BPF.dat file using Qucconv

137MHz_BPF_sfile.sch, test file using 137MHz_BPF.s2p

[image]

[S-parameter_file_test_prj.zip](https://github.com/ra3xdh/qucs_s/files/10871770/S-parameter_file_test_prj.zip)

tomhajjar, 2023-03-02:
3dB_Attenuator files for Qucs-S and ngspice

[3dB_Attenuator_ngspice.zip](https://github.com/ra3xdh/qucs_s/files/10875573/3dB_Attenuator_ngspice.zip)

tomhajjar, 2023-04-22:
There has been some good progress on adding S-parameter file support. I'm hoping the work done by Giles Atkinson and 
Jim Mittel gets fully tested and incorporated in a future release of ngspice. I can't do much until ngspice gets the added support. 

It looks like two things has to happen

1: Update S2spice to support the format required by the future ngspice release. Jim Mittel has modified S2spice for this format. For Windows users, hopefully Dan Dickey can release a new version that gives the option for a "ngspice compatible" file. 

2: A new release of ngspice that supports the file generated by S2spice.

https://sourceforge.net/p/ngspice/discussion/120973/thread/51228e0b01/?page=1

ra3xdh, 2023-04-23:
@tomhajjar Yes, this is good news. I gave advice to the frequency dependent source model developer to submit it to Ngspice upstream. Once it will be available in the Ngspice upstream I will add the S2P device support in the GUI level. It is not a problem to backport the s2spice converter features to the Qucs-S.

transmitterdan, 2023-04-26:
Now I'm confused.  Is Ngspice going to support S-parameter files directly?  Or is it going to support the LTspice/Pspice like frequency dependent sources upon which s2spice depends?

ra3xdh, 2023-04-28:
> Now I'm confused. Is Ngspice going to support S-parameter files directly?

It's better to ask this question at the Ngspice forum (see the link for discussion thread above). Currently the frequency dependent table controlled source still be not finished, so I cannot add it to Qucs-S.

ra3xdh, 2023-05-31:
Yes, I have seen that frequency dependent E-source is preparing for merge into the master branch of the Ngspice. I am planning to start work on the s2p devices support after releasing of the v1.0.3 package (planned in the next 1-2 weeks).

tomhajjar, 2023-05-31:
So this support will preclude having to use S2spice.exe?

ra3xdh, 2023-05-31:
No, I am planning to include s2spice code in the Qucs-S. There will be no need to install this utility separately.

tomhajjar, 2023-05-31:
It hasn't been an issue for me but S-parameters can be in DB or MAG format. In some versions of S2spice I think there was an option to generate the E-sources in various formats.

tomhajjar, 2023-06-01:
I found a possible issue when simulating circuits using more than the 100 sweep points that S2spice defaults to. See attached plots using 100 and 500 points. Problem happens for ANY number greater than 100 points. I'm unsure if this is an interpolation issue or some other problem.

[image]
[image]

[S-parameter_files_ngspice_prj.zip](https://github.com/ra3xdh/qucs_s/files/11619734/S-parameter_files_ngspice_prj.zip)

ra3xdh, 2023-06-01:
> I'm unsure if this is an interpolation issue or some other problem

This is sure not interpolating issue related to Qucs-S plotting engine. I have simulated the auto-generated netlist with CLI version of the Ngspice and obtain exactly the same plot. This issue should be reported to the frequency dependent source developer. 

[image]

tomhajjar, 2023-06-01:
Giles is looking into it.

tomhajjar, 2023-06-09:
I did some additional testing in Qucs. Not sure if this issue has anything to do with issues using Qucs-S and ngspice.

If I generate an S-parameter file of 101 points and then do analysis of the file using 501 points. The Group Delay data has ringing.

If I generate an S-parameter file of 501 points and then do analysis of the file using 501 points. The Group Delay data is OK.

[image]
[image]
[image]

[S-parameter_file_test_prj.zip](https://github.com/ra3xdh/qucs_s/files/11699634/S-parameter_file_test_prj.zip)

ra3xdh, 2023-06-11:
> If I generate an S-parameter file of 101 points and then do analysis of the file using 501 points. The Group Delay data has ringing.

I have checked the provided files. It is not interpolation error of the Qucs/Qucs-S plotting engine. The ringing presents in the dataset and comes from the Qucsator simulation kernel.

tomhajjar, 2023-06-11:
Giles over at ngspice spotted an obvious issue with the Group Delay data. Not sure what the solution is yet. It could be S2spice is doing the conversion wrong.

The Group Delay plots using Qucs for the LC filter and the s-parameter file are the same, 60ns.

The Group Delay plots using ngpice via Qucs-S or DuSpice for the LC filter are the same as Qucs, 60ns.

The Group Delay plots using ngpice via Qucs-S or DuSpice the s-parameter file are 600ns, 10x too high.

[image]

[image]

tomhajjar, 2023-06-14:
Giles over at ngspice found problems with the version of S2spice I was using. Using the version of S2p he used, the problems with group delay or # of test points are gone. I think we are good ;-)

https://www.dropbox.com/s/ujcyn3a7s0qkmcx/s2p.exe

transmitterdan, 2023-06-15:
I have made several changes but may be incompatible with ngspice.  It turns out that using the E (voltage) source is not efficient compared to the G (current) source.  E sources require many more nodes in the admittance matrix than G sources.  So I have elected to create the subckt file with G sources.  This speeds up LTspice simulations about 5:1 for large multi-port networks.

Have a look at the latest here: https://github.com/transmitterdan/s2spice/releases

Would be interesting to understand how the number of points matters.  The code in s2spice does not have a fixed number of points.  It dynamically sizes the arrays as it reads the file.

ra3xdh, 2023-06-15:
@transmitterdan I am planning to start work on s2p files support since the next release of Qucs-S. I considered to add dependency to s2spice utility as the git submodule. But this utility depends on wxWidgets and has GUI-only operation that doesn't fit to concept of the integration in Qucs-S. So I decided to port the `SObject.cpp` to Qt  and use it in the Qucs-S which is not violation of the GPL license. Should I use the state before the 2.1.0 release of s2spice because Ngspice doesn't support frequency-dependent G-sources?

ra3xdh, 2023-06-15:
No, my question is not related to work on frequency dependent E-source support that is done by Giles for Ngspice.

tomhajjar, 2023-06-15:
@transmitterdan

I tested s2spice-2.1.1-win32.exe and it seems to work.

What was the rational for changing the suffix to .inc? It's not that common in LTspice and it's not a default in Qucs-S. 

I realize your target audience is the LTspice user and the suffix is not material.

transmitterdan, 2023-06-15:
> @transmitterdan I am planning to start work on s2p files support since the next release of Qucs-S. I considered to add dependency to s2spice utility as the git submodule. But this utility depends on [...]

Yes, but have a look at the git commit trail after that release to see if any other bugs were fixed after the change to G-source.

G-source is more computationally efficient because the node count is more or less N-squared where N is the number of ports.  But with E sources the node count goes as N-cubed.  For larger devices with many ports (like 8-way splitter/combiner) the memory and execution speed slows down.  It also improves numerical stability.

transmitterdan, 2023-06-16:
> @transmitterdan
> 
> I tested s2spice-2.1.1-win32.exe and it seems to work.
> 
> What was the rational for changing the suffix to .inc? It's not that common in LTspice and it's not a default in Qucs-S.
> 
> I realize your target audience is the LTspice user and the suffix is not material.

Some email systems are refusing to transmit files with name ending in .lib.  The ending .inc works for LTspice and the email systems don't complain about it.

transmitterdan, 2023-06-16:
> @transmitterdan I am planning to start work on s2p files support since the next release of Qucs-S. I considered to add dependency to s2spice utility as the git submodule. But this utility depends on [...]

The program does have a command line interface so it can be run without opening any windows.

tomhajjar, 2023-06-16:
I have tested three variants of the latest versions of "S2spice" by the following authors:
 
Jim Mittel, @jmittel
Giles Atkinson, @gatk555
Dan Dickey, @dldickey

Jim's version uses A sources. Jim says his is a variation of what Giles created but Giles latest rev uses S sources. I'm a bit confused. 

Below is a snippet of the resultant lib/inc files. Attached is the latest versions as of 6/16/23.

Jim

*S11 FREQ R_I
A11 %vd(10 3) %vd(11 12) xfer1
.model xfer1 xfer R_I=true table=[
+ 1.070000e+08Hz 9.106128e-01 -4.131694e-01

Giles

*S11 FREQ R_I
E11 11 12 FREQ {V(10,3)}= R_I
+( 1.070000e+08Hz, 9.106128e-01, -4.131694e-01)

Dan

S11 FREQ R_I
G0101 3 100 FREQ {V(100,3)}= R_I
+( 1.070000e+08Hz, 9.106128e-03, -4.131694e-03)

[S2spice_SW.zip](https://github.com/ra3xdh/qucs_s/files/11772103/S2spice_SW.zip)

tomhajjar, 2023-06-16:
> What are the results? All he same? Or different, so more work needed?

Results seem to be the same. I have been testing using the same files. Jim asked me to test S-parameter files in polar format. Qucs only generates rectangular format so I need to test more files.

wsteffe, 2023-06-22:
I think you may use the vector fitting capabilities of python python module scikit-rf to generate a spice circuit from a touchstone file  (see at: https://scikit-rf.readthedocs.io/en/latest/examples/vectorfitting/vectorfitting_ex2_190ghz_active.html).
The advantage of this approach is that the generated circuit is made of constant lumped elements (not frequency dependent) and it is better suited also for time domain simulations.

tomhajjar, 2023-06-22:
Majority of Spice files are for active devices with gain....

wsteffe, 2023-06-22:
Vector fitting can be applied also to S parameters of active devices. The only difference is that poles of S matrix can stay on right half of the Laplace plane and you must not enforce passivity. Example linked in my previous post is indeed related with an active device. Of course the frequency response of S parameters is valid for small signal regime with a fixed polarization point. 
Anway I thought that this issue was to aimed to the utilization of distributed circuital models (like MStrip elements ..) which were available in the old qucs releases but dismissed in qucs-s. If that was the purpose you may compute the frequeny response using old qcsactor (I am not sure but I think this is the name of the code aimed to this computation) and convert them into general spice circuit with vector fitting.

ra3xdh, 2023-10-06:
I have added s2p file support for Qucs-S. The implementation is based on the code provided by Jim Mittel. This feature will be available since the next release.
[image]

ra3xdh, 2023-10-06:
Closing ticket as completed. The s2p files support will be available since the next release.

closed 2023-10-06 by ra3xdh

oe2lsp, 2023-12-04:
Not working here.
I either get an error massage or qucs-s terminates without any error message.

Qucs-s Version 2.1.0
Ngspice Version 41

two files attached:
[terminate.zip](https://github.com/ra3xdh/qucs_s/files/13546465/terminate.zip)
[error.zip](https://github.com/ra3xdh/qucs_s/files/13546491/error.zip)

Both touchstone files are working fine in qucs, the schematic files were created directly on qucs-s.
If there is any additional information needed, please let me know