ra3xdh/qucs_s #234: SPICE SIMULATION ERROR

sebastianhenaosanta, 2023-02-24:
This is more like a question. I'm now trying to use this tools but i'm getting an issue when trying to run the simulation. 
[image]

If anyone could help, i'll apresiated.

ra3xdh, 2023-02-25:
The resistor model inside the SPICE subcircuit has the wrong syntax. Ngspice accepts R but not RES. See the chapter 3.3.2 of the Ngspice manual for more details.

ra3xdh, 2023-02-25:
See also chapter 3.3.5 of the Ngspice manual for valid resistor model example.

sebastianhenaosanta, 2023-02-25:
Thank you very much,  now is working as you show in the video. I'm not familiarized with spice files. 
 As you said RES is not accepted by NGSPICE and this is what i dit:

*****************************************************************************
* TL431 MACROMODEL ***************3-26-92************************************
* REV N/A ****************************************************************DBB
*****************************************************************************
*              REFERENCE
*              |  ANODE
*              |  |  CATHODE
*              |  |  |
.SUBCKT  TL431 1  2  3
V1  6  7  DC  1.4V
I1  2  4  1E-3
R1  1  2  1.2E6
R2  4  2  RMOD 2.495E3
R3  5  7  .2
D1  3  6  DMOD1
D2  2  3  DMOD1
D3  2  7  DMOD2
E1  5  2  POLY(2)  (4,2)  (1,2)  0  710  -710
.MODEL RMOD **R** (TC1=1.4E-5 TC2=-1E-6)
.MODEL DMOD1 D (RS=.3)
.MODEL DMOD2 D (RS=1E-6)
.ENDS

just change RES by R and now is working

this is the original model:

*****************************************************************************
* TL431 MACROMODEL ***************3-26-92************************************
* REV N/A ****************************************************************DBB
*****************************************************************************
*              REFERENCE
*              |  ANODE
*              |  |  CATHODE
*              |  |  |
.SUBCKT  TL431 1  2  3
V1  6  7  DC  1.4V
I1  2  4  1E-3
R1  1  2  1.2E6
R2  4  2  RMOD 2.495E3
R3  5  7  .2
D1  3  6  DMOD1
D2  2  3  DMOD1
D3  2  7  DMOD2
E1  5  2  POLY(2)  (4,2)  (1,2)  0  710  -710
.MODEL RMOD RES (TC1=1.4E-5 TC2=-1E-6)
.MODEL DMOD1 D (RS=.3)
.MODEL DMOD2 D (RS=1E-6)
.ENDS

ra3xdh, 2023-02-25:
Please note that TL431 model is presented in Regulators library. You don't need to add it manually. Use the Search text field to find library devices.

sebastianhenaosanta, 2023-02-25:
Ok, i'll search all components and models that you have implemented.

ra3xdh, 2023-02-25:
Closing this issue as resolved.

closed 2023-02-25 by ra3xdh