ra3xdh/qucs_s #235: [ REQUEST ] embed simulation console to GUI

dsm, 2023-03-02:
can we embed simulation console output to gui. Every time press exit for if we want to change values on schematic. this thing is so annoying. this is low priority.

ra3xdh, 2023-03-02:
Yes, it is technically possible. I was also considering switching the simulator log output to the main window bottom dock. It should look like the OpenVAF dock here: https://user-images.githubusercontent.com/4920080/208481512-1e38b14a-0655-4036-90f0-b7d8e16b1a48.png

dsm, 2023-03-02:
Great to hear that, it's so good

dsm, 2023-03-02:
maybe we add simulation backend quick selection menu to simulation dock window. when you push to openvaf dock code ?

zergud, 2023-03-03:
what about multiple schematic?
how to separate the results?
whether to lock schematic during simulation (due non-modal panel)?
need to design

ra3xdh, 2023-03-03:
> what about multiple schematic?

I don't see an issue if only the log for the latest simulation is shown in the dock. The saving of the individual logs for every schematic is also technically possible, but it will require a more deep redesign of the application.

ra3xdh, 2023-03-03:
> maybe we add simulation backend quick selection menu to simulation dock window

It remains a lot of incompatible device for each simulation engine. Also I didn't find a way to reload device list in the left panel without application restart. It's better to keep the simulator selection in the application settings. Otherwise it would be hard to prevent user from insertion of the incompatible device and getting hard to debug error message.

denizzzka, 2023-07-18:
> what about multiple schematic?

IMHO, it would be convenient to create checkbox for elements which will be excluded from simulation if this current circuit will be used as subcircuit somewhere else. Such elements could be shown in a more faded or gray color on circuit.

So, it will be possible to test and measure small piece of circuit and then add this to more general circuit

ra3xdh, 2024-08-24:
See also #904. Need to consider if the adding progress bar is possible as part of the future simulation dock.

ra3xdh, 2026-10-30:
Implemented by #1730

closed 2026-10-30 by ra3xdh