ra3xdh/qucs_s #239: Can't set Simulation=tran in Nutmeg formula

bobu01, 2023-03-05:
In qucs-s using the ngspice sim, I'm unable to configure Nutmeg formulas.  Simulation=tran cannot be used. It is stuck at Simulation=ac.  No error was reported. The OS is EndeavorOS (Arch Linux based). qucs-s version 1.0.1-1 was built from AUR repository. Maybe something went wrong in this build.

Here's a schematic from the qucs-s documents, figure 6.2. 
Simulation=tran can be selected in properties, but it won't apply.
[eqn_example.sch.zip](https://github.com/ra3xdh/qucs_s/files/10890555/eqn_example.sch.zip)

Thanks in advance.

ra3xdh, 2023-03-05:
This bug appears only in Qt6 version. Qt5 build is not affected. And Arch has recently switched Qucs-S to Qt6. The `QComboBox::activated()` signal is deprecated in Qt6 and all its occurrence need to be rewritten. All components having properties selectable from combo box are affected. I will try to provide a fix in the next version.

ra3xdh, 2023-03-05:
Only `QComboBox::activated(QString )` need to be fixed

ra3xdh, 2023-03-05:
I have fixed deprecated `activate()` signal. The fix is expected for v1.0.2 release package. You may either to compile application with Qt5 or clone the latest version from Git and compile with Qt6 until release package v1.0.2 will be available. I am closing this ticket as resolved. 

Also please note that Ngspice has introduced a modern syntax for currents in the recent versions. You need to write `i(V1)` instead of `V1#branch` in the equations get current flowing through V1 source. Qucs-S documentation was not updated to reflect this subject.

closed 2023-03-05 by ra3xdh