ra3xdh/qucs_s #244: The program crashes and then closes every time I try to open a file, save something, click on the projects tab or change the location of the ngspice simulation engine

DavidCValbuena, 2023-03-12:
The program crashes and then closes every time I try to open a file, save something, click on the projects tab or change the location of the ngspice simulation engine and I don't know what to do at first it didn't happen to me

ra3xdh, 2023-03-12:
I cannot reproduce the crash using and the portable package v1.0.1. Everything works as expected on four different physical machines using Windows 10 and 11. I need more details to understand the reason of the crashes. I can only guess that you are somehow replaced the Qt5 related DLLs in the installation directory or have corrupt binaries.

ra3xdh, 2023-03-12:
Also make sure that you have downloaded a 64bit package. There exists 32bit package that is not recommended for 64 bit system.

ra3xdh, 2023-04-08:
Still cannot reproduce this. I have tried the release package on different Windows machines and everything works as expected. I can only guess DLLs corruption or mixing DLLs versions. Closing as cannot reproduce.

closed 2023-04-08 by ra3xdh