ra3xdh/qucs_s #247: Trying to run old examples in Qucs-s 0.0.20: Warning: TSTEP is invalid and other issues

JohnCC330, 2023-03-16:
I'm running on a Linux Slackware machine. When I run qucs-s 0.0.20, and I load, for example  [boostconverter.sch](https://qucs.sourceforge.net/examples/boostconverter.sch) (I choose this one because it was similar to what I was trying to do).

The complete response to running this circuit:

```
Ngspice started...
Warning: TSTEP is invalid, must be greater than zero.

TSTOP is invalid, must be greater than zero.

 Error: unknown parameter on .tran - ignored

tran simulation(s) aborted
Error(parse.c--checkvalid): ctrl: no such vector.

Circuit: * qucs 0.0.20 

```

Same happens when when I run the AppImage of the latest qucs-s.

I'm an infrequent user of qucs(-s). Can anyone suggest how to proceed?

ra3xdh, 2023-03-16:
This example is not compatible with Ngspice. The examples distributed with Qucs-S were revised recently and incompatible examples were excluded. See also #162 Also I would advice you don't use any examples or documentation from the Qucs website. Some things may work differently in Qucs and Qucs-S+Ngspice. 

The boost converter example contains parametrized TRAN simulation. Ngspice doesn't support variables as simulation parameters and requires explicit simulation parameters specifications. In other words only numbers are allowed for transient simulation parameters. Also don't forget to specify the "initialDC=no" option because the circuit contains initial conditions. After making necessary corrections the circuit starts to operate. 

[image]

ra3xdh, 2023-03-25:
No further questions from the issue reporter. Assume the problem is resolved. Closing this.

closed 2023-03-25 by ra3xdh