ra3xdh/qucs_s #248: Optocoupler simulation

cmat21, 2023-03-17:
Hello, I am trying to use the spice model of the VO617 optocoupler but the nodes of the spice net do not appear, nor does the ctr model parameter appear. Could you help me to simulate with this model?
Thanks in advance
[BJT_5mA_V2.txt](https://github.com/ra3xdh/qucs_s/files/10997344/BJT_5mA_V2.txt)
[image]

ra3xdh, 2023-03-17:
It is not a bug. Your file has the broken encoding. The most probably it is UTF-8. It could be seen using hexdump. Every symbol has 2 bytes. Ngspice cannot operate with UTF-16 files and Qucs-S doesn't recognize such model. Everything works as expected after I converted the encoding to UTF-8.
~~~
$ hexdump -C  ~/Downloads/BJT_5mA_V2.txt | head -5
00000000  2a 00 20 00 4c 00 69 00  62 00 72 00 61 00 72 00  |*. .L.i.b.r.a.r.|
00000010  79 00 20 00 6f 00 66 00  20 00 44 00 43 00 20 00  |y. .o.f. .D.C. .|
00000020  49 00 6e 00 70 00 75 00  74 00 20 00 50 00 68 00  |I.n.p.u.t. .P.h.|
00000030  6f 00 74 00 6f 00 74 00  72 00 61 00 6e 00 73 00  |o.t.o.t.r.a.n.s.|
00000040  69 00 73 00 74 00 6f 00  72 00 20 00 4f 00 75 00  |i.s.t.o.r. .O.u.|
~~~

cmat21, 2023-03-17:
I have already converted the encoding to UTF-8 using note++. Inside the file there is a parameter declaration "PARAMS: REL_CTR=1" but I don't get the parameter to modify it. Also the simulation results seem not to be within the expected range. Do you have any documentation I can follow? 
[image]
[opto_question.zip](https://github.com/ra3xdh/qucs_s/files/11001182/opto_question.zip)

ra3xdh, 2023-03-17:
The usage of SPICE models is only documented here: https://qucs-s-help.readthedocs.io/en/latest/SubLib.html#using-manufacturers-component-data-libraries 

> https://qucs-s-help.readthedocs.io/en/latest/SubLib.html#using-manufacturers-component-data-libraries

The problem may be with the G-type source with the TABLE dependency. I am not sure if Ngspice fully supports it.

cmat21, 2023-03-18:
Hello @ra3xdh @tomhajjar I thank you for the answers to my query. The models you shared are helpful, QUCS-S is great software. Please continue to make it grow.

closed 2023-03-18 by cmat21