ra3xdh/qucs_s #249: Parametrising square wave source

0range0range, 2023-03-24:
When I try to set the  TL parameter of the square wave source from spice `.param` section, it does not work properly. But other things like amplitude do. Also in case of sine wave it is working. Am I doing something wrong?

QUCS-s version: 1.0.1, OS: Linux.

[image]

ra3xdh, 2023-03-24:
It is not a bug. The square wave source parameters are different from SPICE specification. SPICE uses period and pulse width but Qucs used TL and TH. The period and pulse width are calculated from TL and TH to form a valid SPICE entry for this source. And the calculation is possible only if the TL and TH are defined as numbers. The parameter definition will not work with this source. You need to use a full SPICE definition as the workaround. Look at the attached screenshot.

[image]

0range0range, 2023-03-25:
Worked. Thanks.

closed 2023-03-25 by 0range0range