ra3xdh/qucs_s #250: Setting graph limits manually

0range0range, 2023-03-24:
1. When I try to change limits of the graph manually and previous automatic value is an integer, I am not able to type decimal separators. Neither `.`, not `e-`. I can only put comma, which is being ignored by Qucs-s.

2. Weirdly, if the cell already has a number with fractional part (either in normal or scientific form), then everything is working without problems.

3. Now if I erase, for example, `step` cell completely I can not input fractional numbers. Even though before I cleared it (assuming there was a number with fractional part) I could do it, as explained in 1,2.  Copy-paste of fractional number also does not work

[image]

ra3xdh, 2023-03-24:
What is your system locale? I tried to reproduce this issue with the `en_US` and `de_DE` locale but with no success. The `de_DE` locale begins to accept comma `,` as the delimiter but still accept dot `.`. But I believe the reported issue could take place. The `QDoubleValidator` class has the locate setting method `setLocale()` and the double number syntax may become locale dependent. It seems the validators require an explicit setting of the locale.

ra3xdh, 2023-03-25:
@0range0range , you may try the `current` branch with the latest commit. I have added explicit locale setting for `QDoubleValidator`. But I have tried different locale combination and still not able to reproduce this problem.

ra3xdh, 2023-04-08:
@0range0range Have you tried the `current` branch or setting `en_US` locale like this?
~~~
LANG=en_US qucs-s
~~~
What is your distribution and Qt version?

ra3xdh, 2023-04-23:
No further info regarding this issue. Still cannot reproduce it. Assume this was related to `QValidator` locale setting. Now all `QValidator` instances should be locale-independent. Closing this.

closed 2023-04-23 by ra3xdh