ra3xdh/qucs_s #251: Trying digital components

0range0range, 2023-03-25:
Hello, 

I was trying to play around with some digital components and I could not get it to work at first. 
I had an error: 
```
ERROR - Output delay <= 0 not allowed - output ignored!
  Instance: ay1
  Node: out1
  Time: 0.001000 
```

After reading: https://github.com/ra3xdh/qucs_s/issues/97 , I noticed a `t` parameter for digital blocks and setting it to `1ns` was enough to overcome the error.

1. But now I wonder why does it work and what is the role of `t`? 

2. Wouldn't it be better to set it to 1n by default (as for example for rise and fall time for square wave source) so new people do not have to deal with it when they just want to try 1-2 gates?

3. I tried to change `Voltage of high level` parameter, but it always outputs 3.3V. How does this one works?

Here is the circuit example
[image]

ra3xdh, 2023-03-25:
> But now I wonder why does it work and what is the role of t?

The `t` represents the delay. See here for more information https://ngspice.sourceforge.io/docs/ngspice-html-manual/manual.xhtml#magicparlabel-89763 Existing Qucs symbol doesn't have the separate property for rise and fall delay and assumes t=tr=tf. Otherwise it would require the redesign of the symbol and the loss of the backward compatibility. 

> Wouldn't it be better to set it to 1n by default

Yes, this could be done. The XSPICE models don't support zero delay value. 

> I tried to change Voltage of high level parameter, but it always outputs 3.3V.

This property has no effects with XSPICE devices. You need to define the `vcc` parameter that is used as the VCC voltage of the logic gate devices.
[image]

ra3xdh, 2023-03-25:
I have changed default `t` value to prevent errors coming from XSPICE.

tomhajjar, 2023-03-26:
None of the digital examples in Qucs can be used in Qucs-S. 

Will additional digital components and sources available in Qucs be available in Qucs-S?

ra3xdh, 2023-03-26:
@tomhajjar Currently only logic gates are supported by Qucs-S. Yes, I am planning to add support for the missing devices like flip-flops. Adding support for the digital simulation mode with `iverilog` backend is also technically possible. But currently I have a limited time and not able to make a massive update of the application. Please wait. 

After the digital devices will be finished, it will be needed to update the documentation and add the digital simulation tutorial. Qucs tutorial for digital simulation is not valid for Qucs-S.

closed 2023-03-30 by ra3xdh