ra3xdh/qucs_s #252: Tran settings causing ngspice failure

tomhajjar, 2023-03-30:
In testing a new 555 timer model I ran into an issue. The netlists were tested in ngspice via DuSpice and work. 

The tran statement generated by Qucs seems to be the issue

tran statement that works in ngspice
tran 1u 0.05s uic

Qucs-S generated tran statement that fails in ngspice
tran 0.00049505 0.05 0 

This rev of the XSpice 555 behavioral model works over a wide range of voltages. Clyde is working to simplify it.   

[555behav_PWM_10_90.zip](https://github.com/ra3xdh/qucs_s/files/11115050/555behav_PWM_10_90.zip)
[555behav_PWM_OSC.zip](https://github.com/ra3xdh/qucs_s/files/11115055/555behav_PWM_OSC.zip)

tomhajjar, 2023-03-30:
How does one know when to use this option? Is it just trial and error?

ra3xdh, 2023-03-31:
> How does one know when to use this option? Is it just trial and error?

This is obvious from the option name. The `InitialDC=no` should be set if the circuit contains initial conditions. Otherwise the initial DC simulation will override the initial conditions. If the schematic contains no initial conditions the default `InitialDC=yes` should be left. I cannot say why XSPICE 555 timer model from your schematic requires explicit setting of UIC even if no initial conditions applied. Probably there is something wrong in the model design or there exists undocumented Ngspice feature regarding XSPICE. I would be better to ask this the developer of this model at the Nsgpice forum.

tomhajjar, 2023-04-07:
This is the final cut of the 555 Behavioral model by Clyde. It works over a wide range of supply voltages.

[555 Behavioral Model.zip](https://github.com/ra3xdh/qucs_s/files/11182094/555.Behavioral.Model.zip)

ra3xdh, 2023-04-08:
@tomhajjar Thanks for providing device symbol and subcircuit! I will add this model in the system library in the upcoming release.

closed 2023-04-20 by ra3xdh