ra3xdh/qucs_s #254: Add file I/V sources

ra3xdh, 2023-04-03:
Ngspice supports file-based current and voltage sources via XSPICE device type "filesource". It's need to add a GUI entry for this device. The `ifile.cpp` and `vfile.cpp` should be modified to generate SPICE netlist.

cmat21, 2023-04-05:
Hello, I am waiting for the availability of the sources of V&I based on files. Is there another way to simulate the file-based V&I source?

ra3xdh, 2023-04-06:
The file-based sources are not SPICE core devices, but XSPICE devices. These sources require ports definition and modelcard. You cannot define such type using the red voltage source. Please wait an update. I am planning to prepare v.1.0.2 release in 1-2 months and the next release will include file-based sources.

ra3xdh, 2023-04-18:
I have just added the file voltage and current sources. The difference from the old implementation is that Ngspice doesn't support the cubic interpolation and waveform repeat. Here is an example schematic using these sources.
[image]

ra3xdh, 2023-04-18:
The data file should contain the space separated values. The first column is time and the second column is current/voltage. Here is an example of the DAT file contents.

~~~
1e-3 1.0
2e-3 1.3
3e-3 1.8
4e-3 1.5
5e-3 1.6
6e-3 1.4
7e-3 1.2
8e-3 1.15
9e-3 1.0
10e-3 1.05
~~~

cmat21, 2023-04-18:
Great work @ra3xdh, thank you for taking the project forward!

ra3xdh, 2023-04-18:
Closing this ticket as completed. The file sources will be available since v1.0.2.

closed 2023-04-18 by ra3xdh