ra3xdh/qucs_s #255: Example file s_param_wideband_amp.sch -- S Parameters Simulation employs incompatible ports

csrabak, 2023-04-05:
The file in examples directory ```ngspice/s_param_wideband_amp.sch``` although says it is compatible with NGspice >= 37, employs qucs ports P1 and P2 and does not simulate.

If I understand the documentation correctly these ports need to be replaced by the small signal probes as per "Small signal AC S-parameter simulation"?

However, I cannot find them in the present (1.0.1) palette. .

ra3xdh, 2023-04-05:
I have just checked the mentioned example with Qucs-S v1.0.1+Ngspice-39. The schematic is not broken and could be simulated correctly. See the attache screenshot as the proof. Nsgpice team didn't change the port sources syntax between versions 37 and 39, so the schematic doesn't need an update.

The example with the S-probes from the documentation is totally outdated and is not valid for latest Ngspice. I have removed the releated examples from the Qucs-S distribution. The "RF simulation" chapter also need to be removed from the documentation and replaced by proper tutorial. 

Please ensure the following things to make the S-parameter simulation operational

* Ngspice should be >=37 and S-parameter should be enabled in the build options. It should be enabled by default. Your binary package may have this option disable. Try this netlist https://github.com/imr/ngspice/blob/master/examples/sp/sp1.cir in command line mode. If your Ngspice cannot simulate it, you have something broken with Ngspice.
* Make sure that you have specified proper Ngspice package in the application settings.
* Don't open examples from the `/usr/share` directory where you have no write access. Copy example file to home directory and then simulate it. See also #224 

[image]

csrabak, 2023-04-05:
OK, checking all the stuff per your recommendations, and as I was already using NGSpice 40, I double checked to see if the binary had all the settings as needed and moving the file to a word directory I got:

[image]

I have to confess that I cannot find where in qucs-s the "proper Ngspice package in the application settings" can be set.

[image]

In the screen above you can see the two menu entries I surmised could have the application settings, but not avail...
Presently I undestand qucs-s is using "system wide defaults" or known default for the plattform.

If there's another place pls inform!

Thanks !

ra3xdh, 2023-04-06:
> I have to confess that I cannot find where in qucs-s the "proper Ngspice package in the application settings" can be set.

I meant *Simulation->Select default simulator* menu. Here you can specify the `ngspice_con.exe` location for Windows.

ra3xdh, 2023-04-08:
Assume the issue has been resolved. Closing ticket.

closed 2023-04-08 by ra3xdh