ra3xdh/qucs_s #256: Setting an ideal rectangular voltage source with parameters

cmat21, 2023-04-06:
Hello, could you help me to use parameters within the ideal rectangular voltage source. I attach an image with the results of a simulation comparing the configurations of the sources using parameters. I don't get the same results when I use parameter on "TH=pw" and the same goes for "th","tr","tf","tl". Please tell me if I am forgetting something to configure.
Thank you
[image]
[Source_02.zip](https://github.com/ra3xdh/qucs_s/files/11165728/Source_02.zip)

ra3xdh, 2023-04-06:
The question about rectangular source was raised before several times. It is not a bug. See #249 for explanation. Qucs-S netlister calculates period as `PER=TL+TH` and assumes `TH=PW`. The blue source also ignores the variables, because the calculations cannot be applied for such case. This may be not matching with SPICE. I cannot correct these calculations because the backward compatibility with old Qucs circuits may be lost. Use full SPICE definition using the red source as shown in #249 if you need parametrized schematic.

ra3xdh, 2023-04-06:
Closing as duplicate of #249

closed 2023-04-06 by ra3xdh