ra3xdh/qucs_s #258: Checking some ideal functions blocks

cmat21, 2023-04-12:
Hello everyone, i'm checking some ideal functions blocks for a small project simulation. But i can't make them work properly. Please can you help me. thanks in advance
[image]
[Ideal_01.zip](https://github.com/ra3xdh/qucs_s/files/11216772/Ideal_01.zip)

ra3xdh, 2023-04-13:
These devices are represented as the Qucsator netlists that are stored in the library. Qucs-S converts them back to SPICE and uses `qucs2spice.cpp` converter. It supports only a limited subset of the Qucsator syntax and may not support mathematical functions necessary for these blocks operations.  I am aware about `qucs2spice` problems but I am inclined not to fix it, but provide a library containing true SPICE netlists instead. Don't expect a quick fix here because of low demand on this library. I would advice to use B-type SPICE sources instead of these library if you need these block functions. See also https://ngspice.sourceforge.io/docs/ngspice-html-manual/manual.xhtml#magicparlabel-5564

tomhajjar, 2023-08-01:
Vadim

I did a lot of work on the Ideal library back in 2021 so it would work with ngspice. I recreated the schematics of the functions and modified most of them to work with ngspice. In a few cases there are two variations for the same functions. There is still functions like VCO, Integrator, Quantizer etc to modify but not as many as starting from scratch. 

I made a new Ideal library with cleaned up graphics placed in user_lib. It needs to be updated as functions get updated.

[image]
[image]
[image]

[Ideal Library.zip](https://github.com/ra3xdh/qucs_s/files/12223124/Ideal.Library.zip)

tomhajjar, 2023-10-27:
I dug up my work from 2021 on ideal.lib. I have fixes for many of the functions to work with ngspice. I need to convert my test schematics into subcircuits.

I also reordered the library by function and redid the graphics.  user_lib folder holds the ideal.lib used for testing. 

I made 2 versions of LPF1. Either is OK.

I tried to multiple ways to create "Delay". My lack of experience in ngspice lead to using a TLine to create the delay.

A few problems exist:

The Int and Diff test schematics I made using "ideal.lib" have output amplitudes that look wrong.

AP1 and AP2 never worked in Qucs. AP1 and AP2 use "ModelIncludes" which I can't find info on. (Cascading small L and big I doesn't help clarity...)
 
AP1:
<ModelIncludes "HP1.sch.lst" "LP1.sch.lst" "VSub.sch.lst">

AP2:
<ModelIncludes "BP2.sch.lst" "VSub.sch.lst">

Attached is my test folder and two "ideal" test libraries for the "user_lib" folder. I modify these libraries for my experiments. 

[Ideal Library Test_prj.zip](https://github.com/ra3xdh/qucs_s/files/13191282/Ideal.Library.Test_prj.zip)

tomhajjar, 2023-10-27:
The following functions have not been worked on:

SQRT, QNT, LIM, HardLimiter and DIV.

ra3xdh, 2023-10-28:
@tomhajjar Thanks for providing fixed library. I will try to correct the SPICE entries for Ideal devices where possible by the next release (planned in January or February next year).

tomhajjar, 2023-10-29:
There is one difference between ideal.lib from "Qucs" versus Qucs-S. OpAmp uses a different format. Any idea why this was done?

[image]

tomhajjar, 2023-10-30:
Here is the first cut at a new Ideal.lib. A number of functions haven't been worked on yet. 

Qucs-S won't let you name a new library the same name as in C:\....share\qucs-s\library even though it is stored in user_lib. 

I made the mistake of naming the file Ideal_modified.lib and Qucs-S appends "Ideal_modified" like a hundred times throughout the file....

Old
[image]

New
[image]

[Ideal Library Files.zip](https://github.com/ra3xdh/qucs_s/files/13209583/Ideal.Library.Files.zip)

tomhajjar, 2023-11-01:
More ideal functions like VCO, Limiter, Quantizer... could probably be made to work in ngspice if the syntax for the formulas used in the EDD function are corrected. I don't know what changes to make to the formulas.  

As seen in "AnalogueCM.lib issues #152" there are issues that cause formulas to be processed improperly.

tomhajjar, 2024-01-02:
Ideal.lib is finished for now. @Radvall fixed a number of broken models. 21 of the original 22 models now work. Only VCO is missing. 

Attached are the files used to create and test the models and library. 

[image]

[Ideal_Library_projects.zip](https://github.com/ra3xdh/qucs_s/files/13807157/Ideal_Library_projects.zip)

ra3xdh, 2024-01-02:
Fixed by #428

commit 12be1489 (2024-01-02): Merge pull request #428 from Radvall/fix_ideal_library

Fix ideal.lib; fixes #258
closed 2024-01-02 by ra3xdh