ra3xdh/qucs_s #259: Help error came up: /usr/include/c++/12.2.1/bits/stl_vector.h:1206: std::vector<_Tp, _Alloc>::

abdulbadii, 2023-04-13:
Anyone stumbled to this error too:

`/usr/include/c++/12.2.1/bits/stl_vector.h:1206: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::front() [with _Tp = cmGccStyleDependency; _Alloc = std::allocator<cmGccStyleDependency>; reference = cmGccStyleDependency&]: Assertion '!this->empty()' failed
`
How to solve it ?

ra3xdh, 2023-04-13:
I didn't observe this build error using GCC-11 on Ubuntu-22.04. Also no problems for Fedora. What is your platform and Qt version?

ra3xdh, 2023-04-13:
Are you sure you are using GCC but not Clang for build?

abdulbadii, 2023-04-13:
Arch, update gcc, gcc-libs to ver. 12, qt6-tools but no effect
```

[ 13%] Built target spicecomponents
[ 15%] Built target qt3_compat
/usr/include/c++/12.2.1/bits/stl_vector.h:1206: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::front() [with _Tp = cmGccStyleDependency; _Alloc = std::allocator<cmGccStyleDependency>; reference = cmGccStyleDependency&]: Assertion '!this->empty()' failed.
make[2]: *** [qucs/components/CMakeFiles/components.dir/build.make:2992: qucs/components/CMakeFiles/components.dir/depend] Aborted
make[1]: *** [CMakeFiles/Makefile2:573: qucs/components/CMakeFiles/components.dir/all] Error 2
```
updating ngspice.. see might help

ra3xdh, 2023-04-13:
The Ngspice update will not help here. Ngspice is not a build dependency for the Qucs-S and doesn't affect the build process. I cannot suggest how to resolve this error. I have to install Arch into the VM and see what happens. It's strange that it comes from GCC headers itself but not from Qucs-S sources.

abdulbadii, 2023-04-16:
Still error after update qt6-base if want to build the recent qucs with qt6
`$ sudo cmake -DCMAKE_INSTALL_PREFIX=/usr -DWITH_QT6=ON ..`

failed at same point (percent)
```
[ 13%] Built target spicecomponents
[ 13%] Generating moc_q3scrollview.cpp
[ 13%] Generating moc_q3frame.cpp
[ 13%] Building CXX object qucs/qt3_compat/CMakeFiles/qt3_compat.dir/q3frame.cpp.o
[ 13%] Building CXX object qucs/qt3_compat/CMakeFiles/qt3_compat.dir/q3gdict.cpp.o
[ 14%] Building CXX object qucs/qt3_compat/CMakeFiles/qt3_compat.dir/q3gvector.cpp.o
[ 14%] Building CXX object qucs/qt3_compat/CMakeFiles/qt3_compat.dir/q3glist.cpp.o
[ 14%] Building CXX object qucs/qt3_compat/CMakeFiles/qt3_compat.dir/q3scrollview.cpp.o
/home/budi/Downloads/qucs-s/src/qucs-s-1.0.1/qucs/qt3_compat/q3scrollview.cpp: In member function ‘virtual bool Q3ScrollView::eventFilter(QObject*, QEvent*)’:
/home/budi/Downloads/qucs-s/src/qucs-s-1.0.1/qucs/qt3_compat/q3scrollview.cpp:1504:55: warning: ‘QPoint QDropEvent::pos() const’ is deprecated: Use position().toPoint() [-Wdeprecated-declarations]
 1504 |                 QPoint vp = ((QDragMoveEvent*) e)->pos();
      |                             ~~~~~~~~~~~~~~~~~~~~~~~~~~^~
In file included from /usr/include/qt6/QtGui/qtextlayout.h:7,
                 from /usr/include/qt6/QtGui/qabstracttextdocumentlayout.h:9,
                 from /usr/include/qt6/QtGui/QtGui:6,
                 from /home/budi/Downloads/qucs-s/src/qucs-s-1.0.1/qucs/qt3_compat/q3frame.h:45,
                 from /home/budi/Downloads/qucs-s/src/qucs-s-1.0.1/qucs/qt3_compat/q3scrollview.h:45,
                 from /home/budi/Downloads/qucs-s/src/qucs-s-1.0.1/qucs/qt3_compat/q3scrollview.cpp:48:
/usr/include/qt6/QtGui/qevent.h:706:19: note: declared here
  706 |     inline QPoint pos() const { return position().toPoint(); }
      |                   ^~~
/home/budi/Downloads/qucs-s/src/qucs-s-1.0.1/qucs/qt3_compat/q3scrollview.cpp: In member function ‘virtual void Q3ScrollView::viewportMousePressEvent(QMouseEvent*)’:
/home/budi/Downloads/qucs-s/src/qucs-s-1.0.1/qucs/qt3_compat/q3scrollview.cpp:1740:21: warning: ‘QPoint QMouseEvent::globalPos() const’ is deprecated: Use globalPosition() [-Wdeprecated-declarations]
 1740 |         e->globalPos(), e->button(), e->buttons(), e->modifiers());
      |         ~~~~~~~~~~~~^~
/usr/include/qt6/QtGui/qevent.h:216:19: note: declared here
  216 |     inline QPoint globalPos() const { return globalPosition().toPoint(); }
      |                   ^~~~~~~~~
/home/budi/Downloads/qucs-s/src/qucs-s-1.0.1/qucs/qt3_compat/q3scrollview.cpp: In member function ‘virtual void Q3ScrollView::viewportMouseReleaseEvent(QMouseEvent*)’:
/home/budi/Downloads/qucs-s/src/qucs-s-1.0.1/qucs/qt3_compat/q3scrollview.cpp:1757:21: warning: ‘QPoint QMouseEvent::globalPos() const’ is deprecated: Use globalPosition() [-Wdeprecated-declarations]
 1757 |         e->globalPos(), e->button(), e->buttons(), e->modifiers());
      |         ~~~~~~~~~~~~^~
/usr/include/qt6/QtGui/qevent.h:216:19: note: declared here
  216 |     inline QPoint globalPos() const { return globalPosition().toPoint(); }
      |                   ^~~~~~~~~
/home/budi/Downloads/qucs-s/src/qucs-s-1.0.1/qucs/qt3_compat/q3scrollview.cpp: In member function ‘virtual void Q3ScrollView::viewportMouseDoubleClickEvent(QMouseEvent*)’:
/home/budi/Downloads/qucs-s/src/qucs-s-1.0.1/qucs/qt3_compat/q3scrollview.cpp:1774:21: warning: ‘QPoint QMouseEvent::globalPos() const’ is deprecated: Use globalPosition() [-Wdeprecated-declarations]
 1774 |         e->globalPos(), e->button(), e->buttons(), e->modifiers());
      |         ~~~~~~~~~~~~^~
/usr/include/qt6/QtGui/qevent.h:216:19: note: declared here
  216 |     inline QPoint globalPos() const { return globalPosition().toPoint(); }
      |                   ^~~~~~~~~
/home/budi/Downloads/qucs-s/src/qucs-s-1.0.1/qucs/qt3_compat/q3scrollview.cpp: In member function ‘virtual void Q3ScrollView::viewportMouseMoveEvent(QMouseEvent*)’:
/home/budi/Downloads/qucs-s/src/qucs-s-1.0.1/qucs/qt3_compat/q3scrollview.cpp:1791:21: warning: ‘QPoint QMouseEvent::globalPos() const’ is deprecated: Use globalPosition() [-Wdeprecated-declarations]
 1791 |         e->globalPos(), e->button(), e->buttons(), e->modifiers());
      |         ~~~~~~~~~~~~^~
/usr/include/qt6/QtGui/qevent.h:216:19: note: declared here
  216 |     inline QPoint globalPos() const { return globalPosition().toPoint(); }
      |                   ^~~~~~~~~
/home/budi/Downloads/qucs-s/src/qucs-s-1.0.1/qucs/qt3_compat/q3scrollview.cpp: In member function ‘virtual void Q3ScrollView::viewportDragEnterEvent(QDragEnterEvent*)’:
/home/budi/Downloads/qucs-s/src/qucs-s-1.0.1/qucs/qt3_compat/q3scrollview.cpp:1809:49: warning: ‘QPoint QDropEvent::pos() const’ is deprecated: Use position().toPoint() [-Wdeprecated-declarations]
 1809 |     QDragEnterEvent de(viewportToContents(e->pos()),
      |                                           ~~~~~~^~
/usr/include/qt6/QtGui/qevent.h:706:19: note: declared here
  706 |     inline QPoint pos() const { return position().toPoint(); }
      |                   ^~~
/home/budi/Downloads/qucs-s/src/qucs-s-1.0.1/qucs/qt3_compat/q3scrollview.cpp:1811:38: warning: ‘Qt::MouseButtons QDropEvent::mouseButtons() const’ is deprecated: Use buttons() [-Wdeprecated-declarations]
 1811 |                       e->mouseButtons(),e->keyboardModifiers());
      |                       ~~~~~~~~~~~~~~~^~
/usr/include/qt6/QtGui/qevent.h:710:29: note: declared here
[... cut ...]
/usr/include/c++/12.2.1/bits/stl_vector.h:1206: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::front() [with _Tp = cmGccStyleDependency; _Alloc = std::allocator<cmGccStyleDependency>; reference = cmGccStyleDependency&]: Assertion '!this->empty()' failed.
make[2]: *** [qucs/components/CMakeFiles/components.dir/build.make:2992: qucs/components/CMakeFiles/components.dir/depend] Aborted
make[1]: *** [CMakeFiles/Makefile2:573: qucs/components/CMakeFiles/components.dir/all] Error 2
```

could it be the successful Debian build is of qt5

ra3xdh, 2023-04-16:
I have just tried to compile Qucs-S cloned form git `current` branch on Manjaro using GCC-12.2.1 and Qt-6.4.2 and found no errors. The build was successful and application runs as expected. Also no errors observed with Qt5.15.

What is your exact 3-digit Qt6 version? Also you may try to build Qucs-S wit Qt5. Qt5 need to be installed on your machine. You may control the used Qt version by setting `QT_DIR` variable: https://github.com/ra3xdh/qucs_s/issues/190#issuecomment-1448551134 or here: https://github.com/ra3xdh/qucs_s/pull/217#issuecomment-1448332322 You may need to clean the CMake cache before the build.

abdulbadii, 2023-04-16:
Yup seems the cloned most recent built with qt6 as passed 15% now 17%, though need proven more to full

This one the failed miserable: https://github.com/ra3xdh/qucs_s/releases/download/1.0.1/qucs-s-1.0.1.tar.gz

ra3xdh, 2023-04-16:
Side question. Why are you trying to build the binary from the release tarball? There exists a binary qucs-s package for Arch in the AUR https://aur.archlinux.org/packages/qucs-s

abdulbadii, 2023-04-16:
OK so sorry still wrong
Imho, pretty sure the failure not in later source but in
`make` pause-resume capability is position dependent
this's in (esp. my) case of bad H/W quality or condition, being easily interrupted due to power outage 

if the build progression is before 15% it'd fail do resume, if it's above 48%, i.e. the next link phase, it's capable of resuming

will always build, of course, if remove build dir. and recreate it back, and restart cmake again

closed 2023-04-17 by abdulbadii