ra3xdh/qucs_s #266: Please add Z80 and Intel8080 CPUs simulation support

serfcity, 2023-04-28:
Hello. I know that only tkGate of free (as in freedom) software support Verilog (including Z80 and Intel 8080 CPUs). But tkGate will die soon (it's some sad).

ra3xdh, 2023-04-29:
Adding CPUs model is not a priority in the Qucs-S development. I am mainly intending Qucs-S as an application for simulation of the analog circuits including RF.  

Qucs-S doesn't simulate anything by itself, but only provides an interface for the SPICE simulators. The adding of CPUs models should be done on the simulator side first. It is technically possible for Ngspice, see this unfinished project https://github.com/mcusim/MCUSim But this task is time consuming and has low demand. Ngspice also sets itself as the analog simulator. 

I am planning to return digital simulation mode with IcarusVerilog in the next release. See TODO list for the issue #97 This simulation mode was present in the old Qucs in the past. If there exists a CPU Verilog model it would be possible to attach this model to Qucs-S GUI and simulate it with `iverilog`. The development of the CPUs models is not priority in the Qucs-S development. I have a tasks with a higher priority like providing RF microstrip devices models. 

Summarizing all above, don't expect adding CPUs models in the Qucs-S device library in the near future. But if the digital simulation mode will become operational in the next release, you can try to add Verilog model for the CPU manually.

ra3xdh, 2023-05-29:
The digital simulation with Verilog/VHDL #97 will be available since the next release. The application provides models only for basic digital devices. But one may try to add a more complex device using "VHDL/Verilog file" component. I don't consider the extension of the Qucs-S digital devices library by adding devices like CPUs, memories, etc. The digital simulation in Qucs-S have some limitations. Currently it doesn't support digital buses. This will  require an extension of the schematic editor. It's unclear how to implement it. 

Summarizing all above, I don't expect an extension of the digital simulation subsystem and adding CPU models in the next few years. I am closing this issue.

closed 2023-05-29 by ra3xdh