ra3xdh/qucs_s #267: Error: Failed to start simulator!

mplsgrant, 2023-05-03:
I'm getting a "Failed to start simulator!" error when trying to simulate a circuit. I've included relevant details below. 

# Error details
````
Simulation finished
Ngspice error...
````

# Operating system
````
Debian GNU/Linux 11 (bullseye)
````

# Installation method
````
git clone https://github.com/ra3xdh/qucs_s.git
cd qucs_s/
mkdir builddir
cd builddir/
cmake ..  -DCMAKE_INSTALL_PREFIX=/home/user/qucs-s
make
make install
````
# qucs_s.conf
````
[General]
AdmsXmlBinDir=/home/user/qucs-s/bin
AscoBinDir=/home/user/qucs-s/bin
Attribute=#008080
BGColor=#fffae1
Character=#ff00ff
Comment=#a0a0a4
DefaultSimulator=0
Directive=#008080
EditToolbar=true
Editor=qucs
FileToolbar=true
FileTypes=@Invalid()
GraphAntiAliasing=false
IgnoreVersion=false
Integer=#0000ff
Language=
LargeFontSize=16
NgspiceExecutable=ngspice
NodeWiring=0
Nprocs=4
OctaveExecutable=octave
OpenVAFExecutable=openvaf
QucsHomeDir=/home/user/.qucs
Qucsator=qucsator
Real=#800080
RecentDocs=/home/user/qucs-hello-world/hello-world.sch
S4Q_workdir=/home/user/.qucs/spice4qucs
SimParameters=
SpiceOpusExecutable=spiceopus
String=#ff0000
Task=#800000
TextAntiAliasing=false
Type=#800000
ViewToolbar=true
WorkToolbar=true
XyceExecutable=/usr/local/Xyce-Release-6.8.0-OPENSOURCE/bin/Xyce
XyceParExecutable=mpirun -np %p /usr/local/Xyce-Release-6.8.0-OPENMPI-OPENSOURCE/bin/Xyce
appFont="Cantarell,12,-1,5,50,0,0,0,0,0"
compIconsTheme=0
dx=1440
dy=810
font="Cantarell,12,-1,5,50,0,0,0,0,0"
fullTraceName=false
maxUndo=20
panelIconsTheme=0
x=240
y=135

[Paths]
size=0
````

# hello-world.sch
````
<Qucs Schematic 1.0.2>
<Properties>
  <View=0,0,800,800,1,0,0>
  <Grid=10,10,1>
  <DataSet=hello-world.dat>
  <DataDisplay=hello-world.dpl>
  <OpenDisplay=1>
  <Script=hello-world.m>
  <RunScript=0>
  <showFrame=0>
  <FrameText0=Title>
  <FrameText1=Drawn By:>
  <FrameText2=Date:>
  <FrameText3=Revision:>
</Properties>
<Symbol>
</Symbol>
<Components>
  <GND * 1 480 480 0 0 0 0>
  <GND * 1 260 480 0 0 0 0>
  <Vdc V1 1 260 410 18 -26 0 1 "1 V" 1>
  <R R2 1 480 410 15 -26 0 1 "100 Ohm" 1 "26.85" 0 "0.0" 0 "0.0" 0 "26.85" 0 "european" 0>
  <R R1 1 480 310 15 -26 0 1 "50 Ohm" 1 "26.85" 0 "0.0" 0 "0.0" 0 "26.85" 0 "european" 0>
  <.DC DC1 1 310 150 0 41 0 0 "26.85" 0 "0.001" 0 "1 pA" 0 "1 uV" 0 "no" 0 "150" 0 "no" 0 "none" 0 "CroutLU" 0>
</Components>
<Wires>
  <260 440 260 480 "" 0 0 0 "">
  <480 440 480 480 "" 0 0 0 "">
  <480 340 480 380 "divide" 510 330 18 "">
  <260 280 480 280 "" 0 0 0 "">
  <260 280 260 380 "" 0 0 0 "">
</Wires>
<Diagrams>
</Diagrams>
<Paintings>
</Paintings>
````
# ngspice output
````
user@comp:~$ ngspice
******
** ngspice-39 : Circuit level simulation program
** The U. C. Berkeley CAD Group
** Copyright 1985-1994, Regents of the University of California.
** Copyright 2001-2022, The ngspice team.
** Please get your ngspice manual from http://ngspice.sourceforge.net/docs.html
** Please file your bug-reports at http://ngspice.sourceforge.net/bugrep.html
** Creation Date: Tue Feb 14 14:36:24 UTC 2023
******
ngspice 1 -> 
````

ra3xdh, 2023-05-03:
The schematic contains only DC simulation. It has no effect with Ngspice. This was valid for Qucs but is not valid for Qucs-S. The simulation will not start. Don't follow Qucs tutorial when using Qucs-S. A new tutorial is available here: https://ra3xdh.github.io/pdf/qucs_s_tutorial.pdf 

Also there is a binary package for Debian 11. Why are you compiling the application from source? Find the package and installation instructions here: https://software.opensuse.org/download.html?project=home%3Ara3xdh&package=qucs-s

ra3xdh, 2023-05-03:
Qucs-S should show warning if only DC simulation is presented in the schematic. I have just discovered that this warning is not show anymore after #209 I will provide a fix soon.

ra3xdh, 2023-05-03:
I have added a fix for showing of the checker errors on the `current` branch. The fix will be available in the next release. Now Qucs-S should show detailed error message that schematic contains only DC simulation instead of simple "Ngspice error".

mplsgrant, 2023-05-03:
Thank you for the fast response and fix. I appreciate it.

> here is a binary package for Debian 11

I originally chose to use the AppImage which produced the error I mentioned. Then I chose to compile the program myself in case there was an issue with the AppImage.

In both cases, I wanted to install the software locally using only my "user" account and without leveraging sudo which the .deb packages would require.

ra3xdh, 2023-05-04:
Closing as resolved.

closed 2023-05-04 by ra3xdh