ra3xdh/qucs_s #269: Не запускается симуляция схемы

Laimer641, 2023-05-19:
**не запускается симуляция**
[image]
[image]
[image]

**сам симулятор в консоли запускается**
[image]

** настройки путей для симулятора**
[image]

**Не запускается симуляция даже если запустить qucs-s под sudo**
**Система новая. специально установил для чистоты эксперимента. Система стоит на VirtualBox.**

ra3xdh, 2023-05-20:
Это не баг. На схеме присутствует только моделирование "DC analysis". Оно само по себе не имеет эффекта. Чтобы симуляция заработала, следует задать моделирования AC или Transisent. Показ соответствующего предупреждения пользователю будет исправлен в следующей версии. См. также #267 

И здесь я вижу схему из туториала от старого Qucs. Не надо пользоваться старым туториалом. В Qucs-S что-то может работать иначе. Правильный туториал можно взять здесь: https://ra3xdh.github.io/pdf/qucs_s_tutorial.pdf На него же ведут ссылки в меню Help.

ra3xdh, 2023-05-26:
Предполагаею, что вопрос решен, поэтому закрываю тикет. См. также #271

closed 2023-05-26 by ra3xdh