ra3xdh/qucs_s #277: AC analysis at fixed frequency and plot against sweep parameters other than frequency

niuniuaba, 2023-06-03:
Hello there,

I am trying to do an AC analysis at a fixed frequency (250KHz) with DC voltage as a sweep parameter. After the analysis I want to plot something defined by Nutmeg equation against the DC sweep rather than frequency. But no data plotted in the graph and I noticed there is a 'invalid' at the end of the name of Y axis. Did I do anything wrong? (engine is Ngspice).

[schematic and plot is here in png](https://github.com/ra3xdh/qucs_s/assets/11435613/7ebb5949-534d-48c3-af05-e64363f5ed34)

I also attached the netlist and dataset file here. Any help will be greatly appriciated.

[netlist_dataset.zip](https://github.com/ra3xdh/qucs_s/files/11642378/netlist_dataset.zip)

Best regards,
niuniuaba

ra3xdh, 2023-06-03:
I am aware about this problem. The fixed point AC analysis in Ngspice is a hack. And parameter sweep is also implemented in Qucs-S using hack with for loop, because Ngspice doesn't have `.STEP` analysis mode like XYCE. If you are trying to implement AC(1-point) with Par.Sweep in will produce a set of 1x1 point plots instead a smooth curve. Look also here #156 and #161. There is no workaround for this issue. And it still be unclear how to fix it in the application.

holvo, 2023-06-03:
Please have a look at https://ngspice.sourceforge.io/ngspice-control-language-tutorial.html#step for the capability of the control language for loops and .step emulation.

A modified netlist for use in plain ngspice is attached, which enables plotting of ac data versus dc sweep voltage.
[niuniuaba_mod.zip](https://github.com/ra3xdh/qucs_s/files/11644319/niuniuaba_mod.zip)

holvo, 2023-06-04:
These discussions lead to new ideas and options: On ngspice git branch pre-master there is a new control language function cvector, which creates a complex vector, to store ac loop results directly.
[niuniuaba_mod.zip](https://github.com/ra3xdh/qucs_s/files/11645126/niuniuaba_mod.zip)

ra3xdh, 2023-06-04:
Hello Holger, thanks for looking into this. I have added an example of Qucs-S schematic implementing CV-curve plotting based on your netlist example. This example uses "Nitmeg script" simulation mode.  @niuniuaba , I hope this may help you. I will include this example into the Qucs-S distribution. This may be help for other users trying to plot CV-curve.

The attached screenshots show simulation example and plot settings. 

[image]

[image]

[image]

ra3xdh, 2023-06-04:
I have uploaded CV-curve simulation example schematic in the recent commit.

niuniuaba, 2023-06-06:
Hello @ra3xdh and @holvo 

Thank you very much for you help. Now I can plot CV curve in my case.
It's a bit off topic but I noticed that there is kind of difference in the curve from that of LTspice if I use tenary_fcn explicitly but if I use if() function and set ngbehavior=ltpsa I get almost same curve as LTspice.
One more question, is it possible for qucs-s to read local .spiceinit file when it invokes ngspice binary so that one doesn't have to put a .spiceinit section in every schematic?

[image]

[image]

Best regards,
niuniuaba

ra3xdh, 2023-06-06:
> if I use if() function and set ngbehavior=ltpsa I get almost same curve as LTspice

This should be reported to Ngspice team. Please provide a minimal example that can reproduce the different behavior. This may be not directly to `if` operator, but some other fucntions may work differently if compatibility mode is applied.

>  is it possible for qucs-s to read local .spiceinit file when it invokes ngspice binary

No, there is no way to set the global `.spiceinit` from the Qucs-S GUI. You should include the settings in schematic.

ra3xdh, 2023-06-06:
Closing this as resolved.

closed 2023-06-06 by ra3xdh