ra3xdh/qucs_s #279: Need for "ngspice_mathfunc.inc" in ngspice netlist

tomhajjar, 2023-06-07:
An issue came up in the ngspice forum. What is the reason for .INCLUDE....ngspice_mathfunc.inc in the netlist? It causes issues for those using ngspice to analyze netlists I send them. DuSpice however doesn't seem to care if the line is commented out or deleted. 

.INCLUDE "C:/QUCS-S 1.0.2/share/qucs-s/xspice_cmlib/include/ngspice_mathfunc.inc"

To make files usable for the ngspice forum I edit them as attached. I comment out a bunch of lines and add plot statements.

[S-parameters.zip](https://github.com/ra3xdh/qucs_s/files/11679103/S-parameters.zip)

ra3xdh, 2023-06-07:
This file defines some functions needed for compact modelling like `limexp()`. These functions are not needed for 90% cases. Simple delete the related line when sending the auto-generated netlist outside. This file contains nothing related to S-aprameter simulation and this include could be deleted.

ra3xdh, 2023-06-13:
Assume the question about the purpose of `ngspice_mathfunc.inc` answered. Closing this.

closed 2023-06-13 by ra3xdh