ra3xdh/qucs_s #286: Questions about _netN node name , Insert images and ground connection

cmat21, 2023-06-21:
Hello, is there a way to preserve all the default names of the "_netN node" automatically assigned to the circuit, when it is only of interest to put a name to a "net node" in particular using the "wire label"? 
Are you planning to insert images to the schematic sheet from the "paintings" menu?
And, is there a way to wire ground in a shorter way?
[image]

ra3xdh, 2023-06-22:
> is there a way to preserve all the default names of the "_netN node" automatically assigned to the circuit

No, there is no way to control automatically assigned internal node names. I don't plan to add this feature. Use explicit node naming with "wire label" if you need to plot simulation on graphs. Don't rely on implicit node names. 

> Are you planning to insert images to the schematic sheet from the "paintings" menu

It may be technically possible, bit I don't plan to add "Raster image" painting type because of low demand. Export schematic to PNG/SVG and use graphics editor if it's need to combine schematic and image.

> And, is there a way to wire ground in a shorter way

I didn't understand this question. The ground may be connected at any wire. And you can connect ground directly to device pins. Look at examples at `share/examples` directory.

ra3xdh, 2023-06-24:
No further questions from issue reporter. Assume the question is cleared. Closing this.

closed 2023-06-24 by ra3xdh