ra3xdh/qucs_s #287: S parameter simulation

wsteffe, 2023-06-24:
I have just tried to use qucs-s using the filter example given at https://github.com/ra3xdh/qucs_s/issues/233.
I have noticed that S parameter block doesn't work with spice simulators (ngspice or xyce) so I had to build and install qucsator to compute it.
I would like to have the possibility to use the "S parameter simulation" block also with spice simulators. AC analysis of spice simulators is close to that but a little bit different. You have to make separate simulations with different excitation sources and assemble all the results to get the scattering matrix of a multiport device. Would it possible to automate this process to get the result of "S parameter simulation" block with qucsator ?
It would be even better if spice simulators were able to compute directly the solutions associated with multiple exitations. This would save time because (in case of a direct solver) a single LU decomposition could be used for all solutions.

ra3xdh, 2023-06-24:
S-parameter simulation works using the Ngspice >= 37. It is implemented since Qucs-S v0.0.24. I am regularly testing it using Ubunut and WIndows. What Ngspice version are you using?

ra3xdh, 2023-06-24:
Have you tried to simulate the schematic from `share/examples/ngspice/s_param_wideband.sch` ? It should operate properly using Ngspice >= 37.

ra3xdh, 2023-06-24:
Also keep in mind that Qucs equations containing S-parameter will not work with Ngspice. You have to use Nutmeg syntax instead.

ra3xdh, 2023-06-24:
See also #84

ra3xdh, 2023-06-24:
Also you have to build Ngspice from Git `pre-master` branch to simulate schematic from #233 The block representing s2p-defined device contains frequency-dependent E and G sources that are not available in Ngspice release version.

wsteffe, 2023-06-24:
Ok I see. In fact I was using ngspice 36 which is included ubuntu 22.04 distribution. I have just started using qucs-s. Yesterday I built last version of xycfe and trilinos.  I will do the same with Ngspice from git as you suggested. 

Then there is the equations problem. I was thinking that equations and nutmeg are the same. I need to learn something more on the qucs.

wsteffe, 2023-06-24:
I have just bult and installed ngspice from github and I would report an annoying problem: I wanted to remove the older ngspice but when I do that apt want to remove also  qucs-s is installed from http://download.opensuse.org/repositories/home:/ra3xdh/xUbuntu_22.04/ /.

tomhajjar, 2023-06-24:
Attached is a project with S-parameter examples. It should answer many of your questions.

By coincidence a few Qucs-S and ngspice math functions share the same syntax. This is due to ngspice accepting shortened names that happen to correspond with Qucs-S function names. There are old Qucs and Qucs-S files using legacy equations that work. It is pure luck however to have this happen. You should use Nutmeg syntax to avoid issues.

Below is the few math functions that work in both Qucs and ngspice. 

"mag"="magnitude"

"ph"="phase"

"cph"="cphase"

"real"="re"

"imag"="im"

"log"="ln"   ;natural log 
    
"vecmin"="minimum"

"vecmax"="maximum"

[S-parameter_files_ngspice_prj.zip](https://github.com/ra3xdh/qucs_s/files/11857084/S-parameter_files_ngspice_prj.zip)

ra3xdh, 2023-06-24:
> Then there is the equations problem. 

It is not possible to implement automatic full converter from Qucs equation syntax to Nutmeg syntax. Qucs-S supports only a limited subset of Qucs mathematical functions. You must use Nutmeg equations in new simulations. You may find a list of supported functions in the Qucs-S tutorial here: https://ra3xdh.github.io/pdf/qucs_s_tutorial.pdf For more details refer to the Chapter 17 of the Ngspice manual: https://ngspice.sourceforge.io/docs/ngspice-html-manual/manual.xhtml#magicparlabel-23879

ra3xdh, 2023-06-24:
The problem was related to the wrong Ngspice version. Closing this as resolved.

closed 2023-06-24 by ra3xdh

wsteffe, 2023-06-24:
I have stiil problems also using the project provided by tomhajjar.
Here are the error messages reported after simulation:

> Ngspice started...
> 
> ngspice stopped due to error, no simulation run!
> 
> ERROR: fatal error in ngspice, exit(1)
> 
> Circuit: * qucs 1.1.0 /home/walter/.qucs/s-parameter_files_ngspice_prj/137mhz_bpf.sch
> 
> Error on line 4 :
>   vp2 _net0 0 dc 0 ac 0.316228 sin(0 0.316228 1g) portnum 2 z0 50
>  unknown parameter (portnum) 
> Error on line 5 :
>   vp1 _net1 0 dc 0 ac 0.316228 sin(0 0.316228 1g) portnum 1 z0 50
>  unknown parameter (portnum) 
> Simulation finished
>

ra3xdh, 2023-06-24:
It seems you still use system Ngspice-36. Don't forget to update simulator settings. *Simulation->Simulator settings*. Specify the full path to Ngspice binary.
[image]

wsteffe, 2023-06-24:
I have done that but, probably I am using an old ngspice for  another reason.
I have just looked at github project https://github.com/ngspice/ngspice and I have seen that the master branch was lastly updated 8 years ago. And this is the newest branch. Where have you found a newer version ?

ra3xdh, 2023-06-24:
Ngspice is hosted on Sourceforge. The Github mirror is not maintained. See the following:

* https://sourceforge.net/projects/ngspice/
* https://sourceforge.net/p/ngspice/ngspice/ci/master/tree/ You need `pre-master` branch from here.

wsteffe, 2023-06-24:
Thanks. I have built ngspice-premaster from SourceForge (and also qucs-s from github) and now it works.