ra3xdh/qucs_s #288: Issue exporting monochrome image

cmat21, 2023-07-02:
Hi, there may be a problem exporting the schematic as a monochrome image.
[image]
[image]
[image]

ra3xdh, 2023-07-05:
This issue is related to the color mapping while converting image in Qt. I can add a Grayscale option. I hope it may help you.

ra3xdh, 2023-07-05:
I have added a grayscale option in the "Export to image" dialog. This feature will be available since the next release. Closing this.

commit 1c97baa6 (2023-07-05): Add grayscale export image option. Fixed #288
closed 2023-07-05 by ra3xdh