ra3xdh/qucs_s #289: Trouble simulating qucs_s under Windows

lucaaamaral, 2023-07-04:
I've used qucs sumulator before and looked for the latest package, but i'm unable to run any simulation. 
Here is what I have done, correct me if I've done anything wrong:

1- Downloaded the latest release of qucs_s: qucs_s_win64_1_1_0.zip.
2- Downloaded the latest release of ngspice: ngspice-40_64.7z
3- Unzipped the ngspice file and moved the Spice64 folder and it's contents to C:\Spice64
4- Unzipped the qucs_s file and moved the qucs_s_win64 folder and it's content to D:\qucs_s\qucs_s_win64
5- I've ran the qucs executable, draw a simple voltage divider, placed a voltage source and pressed the simulate button and got the error "Failed to start Simulator" followed by a window that showed a "Simulation finished Ngspice error...".
6- Went to the "Simulation > Select default simulator" menu and placed the complete path to the ngspice executable: "C:/Spice64/bin/ngspice.exe". Ngspice was already the default simulator in the qucs_s. I've saved and closed everything and openned up again, but the same result has been seen.

I could not find the qucsator executable in the zip of the release. 
Please tell me if I've done anything wrong in the installation process and what to do to be able to run simmulations again.

lucaaamaral, 2023-07-04:
Sorry, forgot to place a ground to the simulation.

ra3xdh, 2023-07-05:
Yes, this issue was resolved by itself, but I have found a small bug related to this issue. I forgot to fix showing "Missing ground" warning in #267 I have just added a fix for this. 

Some comments related to original issue report:

* It's recommended to use `ngspice_con.exe` instead of `ngspice.exe`. The `ngspice.exe` is the GUI version that will work, but cannot write logs properly.
* Qucsator has been deprecated and may be required only for simulation of microstrip devices. Use new tutorial for Qucs-S: https://ra3xdh.github.io/pdf/qucs_s_tutorial.pdf

closed 2023-07-05 by ra3xdh