ra3xdh/qucs_s #290: Built-in VCO from "Ideal" library is broken?

denizzzka, 2023-07-11:
[image]

```
Ngspice started...
warning, can't find model 'c' from line
    cc1 gnd _net1 c
Netlist line no. 0:
Undefined parameter [v1]
Netlist line no. 0:
Cannot compute substitute

ERROR: fatal error in ngspice, exit(1)

Note: No compatibility mode selected!

Circuit: * qucs 1.1.0 /home/denizzz/.qucs/test_bug_prj/vco_test.sch

Simulation finished
```

ra3xdh, 2023-07-11:
Yes, I am aware about this problem. Furthermore this issue was already reported here #258 Some devices from `Ideal` library still be qucsator only. See my explanations in #258 There is no workaround to put these devices in operation. It's need to edit the library and provide SPICE entries for missing devices. I have planned the revision of the `Ideal` library, but I cannot promise to start this work till the next release.

ra3xdh, 2023-07-11:
Closing this as duplicate. Use #258 to track the status of the `Ideal` library update.

closed 2023-07-11 by ra3xdh