ra3xdh/qucs_s #292: Request: Update translaltion files

antohami, 2023-07-20:
Some dialogues are missing translation. But these phrases are not in the translation files.
I also tried turning on the build option -DUPDATE_TRANSLATIONS=ON
This resulted in an error:
`[ 62%] Generating .lupdate/qucs_ar.ts.stamp
cd /usr/src/RPM/BUILD/qucs-s-1.1.0/x86_64-alt-linux/translations && /usr/share/qt5/bin/lupdate @ -ts /usr/src/RPM/BUILD/qucs-s-1.1.0/translations/qucs_ar.ts
QFSFileEngine::open: No file name specified
lupdate error: List file '' is not readable.
gmake[2]: *** [translations/CMakeFiles/translations.dir/build.make:177: translations/.lupdate/qucs_ar.ts.stamp] Error 1
gmake[2]: Leaving directory '/usr/src/RPM/BUILD/qucs-s-1.1.0/x86_64-alt-linux'
gmake[1]: *** [CMakeFiles/Makefile2:896: translations/CMakeFiles/translations.dir/all] Error 2 `
[image]
[image]
[image]

antohami, 2023-07-28:
The issue has been fixed in the `current` branch. It is possible to update translations.

closed 2023-10-17 by ra3xdh

ra3xdh, 2023-10-17:
Russian translations has been updated by #324. Native speaker required for other languages. Closing for now as completed.

PR #324 (2023-10-17): Update Russian translation
Fixes #292