ra3xdh/qucs_s #293: Empty subcircuits outputs equal to zero

denizzzka, 2023-07-21:
For ease of circuit debugging, I made such stub subcircuit and figured out what it doesn't works as expected without additional zero resistors.

[image]
[image]

It was too hard to understand what goes wrong!

ra3xdh, 2023-07-22:
It is not a bug. The short circuit between the subcircuit ports is not allowed in the Ngspice netlist syntax. You have to add zero-Ohm resistors to obtain short circuit. Open subcircuit ports are in principle allowed.

denizzzka, 2023-07-22:
It is possible to detect this state and display warning?

closed 2025-01-03 by ra3xdh